RavenSystem/esp-homekit-devices #384: Reset Issues

ScottLeMagicien, 2019-05-13:
I've got some Shelly 1's that have been flashed, configured for 12v DC use, and connected to my Home app.   Now I want to switch them to a new network, but I cannot get them to do a Wifi reset not matter what I try.  

I can use them with the Home and the Eve. apps, but I cannot Wifi reset them using the Eve. app since the Show Setup switch does nothing.  (It worked for some of them, but not all of them.)

If I turn off the network they were on, I do see them in AP mode after 5 minutes, but after connecting to them does either nothing happens, or I cannot connect to them at all.

I have tried resetting by switching them on and off 10 times within 3 seconds between 10 seconds after powering up and before 2 minutes have passed.  Nothing.  I have tried to do this with the Home app and various combinations of external switches.  Nothing works.  I was told this should be the external switch hookup:
[image]

Any suggestions would be most appreciated!!

mafyata, 2019-05-13:
I think that “N” is “GND” , “L” is “+VDC”.

ScottLeMagicien, 2019-05-13:
No, that's an old photo.  N is definitely + and L is - on the current model.

ScottLeMagicien, 2019-05-14:
I managed to migrate one Shelly 1 to our new WiFi network after connecting to it's WiFi signal using Safari by typing in the address 192.168.4.1 (Its IPv4 address was automatically set at 192.168.4.2)  But I cannot seem to do that with any of the others, even if I can connect to their WiFi signal.  Setting the IPv4 address, etc. manually also doesn't let me do anything with it in Safari.

ScottLeMagicien, 2019-05-16:
A possible thing we discovered: We had not set the External Toggle 1 setting in the Set Up menu in the Eve. app.  That may be why they didn't respond to the switch when we were trying to reset them...

RavenSystem, 2019-05-16:
@ScottLeMagicien To reset it, you can use external toggles when are activated, or HomeKit Switches from iOS Home App.

But if you want only to change wifi settings, the best is reset it from EVE App, and if your configures wifi are not available, wait for 5 minutes until device enters into AP mode. If you are not able to connect to device AP try removing wifi network from your iOS device, repower your Shelly, and connect to its AP again.

closed 2019-05-26 by RavenSystem

ScottLeMagicien, 2019-06-15:
Dear RavenSystem,

Thanks for your response.  I have not been able to reset a single Shelly 1 no matter what I try :-(  

Is my external switch setup correct?  (In the illustration above?)

And several of our Shelly 1's never enter AP mode when I remove the Wifi Network.  Others show up but I cannot connect to them.  Very frustrating!