RavenSystem/esp-homekit-devices #385: Wired door/window sensor by GPIO

kittmark, 2019-05-13:
Hello, I’ve seen some custom firmwares where the shorting of two GPIO pins using a legacy wired magnetic door sensor is used to simulate an homekit door/window sensor.

This could be useful both for retro-fitting old security systems and for some situations where you will not be able to change batteries to a wireless door sensor (like for anti-tampering of an outdoor box).

Could this type be added to RavenCore? Is the new “type 16” related to this? 
Thanks

RavenSystem, 2019-05-13:
This will be device type 16.

GPL71, 2019-05-14:
@RavenSystem Josè then you develope RavenCore and HAA too?

RavenSystem, 2019-05-14:
@GPL71 yes, but dev type 16 will be the last I will include in RC. And no new dev boards will be included. Then I will only support for bug fixed and stability.

kittmark, 2019-05-14:
Awesome that type 16 will do that, perfect!

lizzus, 2019-05-19:
typ6 16 will be avaible in Ravencore 0.9.0?

RavenSystem, 2019-05-19:
@lizzus Sure.

RavenSystem, 2019-05-19:
```
// Sensors
#define CONTACT_SENSOR                  0
#define MOTION_SENSOR                   1
#define OCCUPANCY_SENSOR                2
#define LEAK_SENSOR                     3
#define SMOKE_SENSOR                    4
#define CARBON_MONOXIDE_SENSOR          5
#define CARBON_DIOXIDE_SENSOR           6
#define FILTER_CHANGE_SENSOR            7
```

mafyata, 2019-05-19:
Can you add:
#define TEMP_HUMIDITY_SENSOR               8

NODEMCU esp8266 + dht22
Now it works great with device type 6(sonoff th) , but i have one switch that is unused.

RavenSystem, 2019-05-19:
@mafyata No, only boolean sensors. And all of them are in the list.

mafyata, 2019-05-19:
It’s a bad news for me. Maybe function to hide switch button :)

RavenSystem, 2019-05-19:
@mafyata You will be able to do that with HAA.

mafyata, 2019-05-19:
Good news. 
It is hard to me to make the correct json options for my devices. I will wait for the software for json configuration.

RavenSystem, 2019-05-19:
@mafyata Don't worry, a JSON for a single accessory with a temp/hum sensor will be very easy. You only must know GPIO and sensor type. Sure that I will write a sample for that here: https://github.com/RavenSystem/esp-homekit-devices/wiki/HAA-Templates

mafyata, 2019-05-19:
Yes , I know them , but don’t know for my other devices - magic home rgb led , sonoff basic new layot and sonoff th 16.

RavenSystem, 2019-05-19:
@mafyata But you can keep using RavenCore in those devices. HAA is not a replacement for RC, and least not in a near future. Maybe, when HAA will do all that RC does, and a JSON generator will be done too.

RavenSystem, 2019-05-26:
RC will not get this finally. HAA will do it.

closed 2019-05-26 by RavenSystem

jat80, 2019-10-01:
> RC will not get this finally. HAA will do it.

Could someone show me how to do this with HAA?  I would like to create a window sensor using a Sonoff SV and a reed switch.