RavenSystem/esp-homekit-devices #386: HAA question

tohmc, 2019-05-14:
In HAA Wiki you explain these:

"t" Type:

Select accessory type:

1: Default. Switch.
2: Outlet.
3: ...
Depending of selected accessory type, there will be different options to configure.

What are the other types?
3,4,5,6... and so on.

RavenSystem, 2019-05-14:
Other types could be water valve, stateless button, lock mechanism, garage door, motion sensor... There is nothing planed yet.

tohmc, 2019-05-14:
A ok.
Its the First step and you added only 1 and 2.

Ok what if u make the Type like the Original UUID2?
Example:
Type „BB“ for air purifier

tohmc, 2019-05-14:
So Everyone can do it on his own. And can select via Type what functions it has and Get an symbol matching to it Function.

RavenSystem, 2019-05-14:
Because “BB” is a char[2] while 1, 2, 3 are numbers. It is a great difference. Compare numbers is easier than compare strings.

greensouth, 2019-05-15:
I've ordered a couple of NodeMCU boards for testing!! Can't wait to learn a few things about this.
Thinking of installing a little step motor to build a watering pump with some 3d printer pieces.

I've been schemes and Arduino IDE code to control the steps of the motor, but I'm a bit lost about using HAA instead. 

I would just choose the right device (one which could let an GPIO active in order to continuosly move the motor) right?

lizzus, 2019-05-15:
I tryed to add follow JSON for a ESP-01
{“c”:{“o”:0,”l”:1,”i”:1},”a”:[{“t”:1,”r”:[{“g”:3}],”b”:[{“g”:0,”t”:1}]}]}
but maybe something is wrong...

RavenSystem, 2019-05-15:
I think problem are the quotes. Try it: {"c":{"o":0,"l":1,"i":1},"a":[{"t":1,"r":[{"g":3}],"b":[{"g":0,"t":1}]}]}

lizzus, 2019-05-15:
OMG.... yes it's true. When i Copy and paste string! Now work! Thanks Josè

skop1986, 2019-05-16:
@RavenSystem 

is it possible to make a web interface for customization, as it is done in tasmota firmware?

P.S. thank you for your work!    what you do is super!

lizzus, 2019-05-16:
Do you mean in captive portale page?

skop1986, 2019-05-16:
@lizzus 

I mean, an inteface like this one.
[image]

RavenSystem, 2019-05-16:
> is it possible to make a web interface for customization, as it is done in tasmota firmware?

@skop1986 https://github.com/RavenSystem/esp-homekit-devices/issues/206#issuecomment-491975743

closed 2019-05-18 by RavenSystem

tohmc, 2019-08-09:
Hi, I have 3 questions.

Could you please add Temperatur/humidity sensor to your define# sensor todo-list?
Maybe DHT-11 or/and DHT-22 support like in ravencore firmware.

Is it possible to add MQTT to your firmware? Or will it have to much space usage?

And last one 
How can I make your firmware to show in HomeKit as a sprinkler or other devices and not the standard switch/light/Ventilator ?