RavenSystem/esp-homekit-devices #387: Shelly 2.5 connected to two wall switches only triggers one light when both are pressed

ivorocha90, 2019-05-15:
Hi,

I've hooked up a Shelly 2.5 with two ceiling lights in the living room. Each Shelly's output is connected to its lamp, and each switch input is connected to its respective wall switch. Both switches are side by side in the wall.

When I toggle each wall switch individually, lights are turned on/off correctly, but if I press both at the same time, only one light is turned on/off. The other one remains in the state it was.

Everything else is working flawlessly.

The setup of the Shelly is the following:
Board type: 5
Device type: 2
External toggle 1: 2
Init state 1: 2
External toggle 2: 2
Init state 2: 2

Thanks.

RavenSystem, 2019-05-15:
Yes, it is normal. When an input is triggered, there is a disable time of 80ms to protect the rest of inputs. It was necessary in older versions, but maybe with my last version of adv-button-library, that disable time of 80ms may not be needed.

ivorocha90, 2019-05-15:
Is there a way to reduce or clear that delay? Maybe with a new parameter for this board/device type allowing to define the amount of the delay?

RavenSystem, 2019-05-15:
Give me some time to test if I can remove it definitely.

RavenSystem, 2019-05-18:
Try version 0.8.11 and comment, please.

ivorocha90, 2019-05-18:
Hi,

Just installed the new firmware version and it is working perfectly. Both lights now turn on/off when pressing both switches simultaneously.

Very appreciated of the work you're doing here.

Thanks.

RavenSystem, 2019-05-18:
Many thanks! Stay tuned, because if for any reason is necessary to disable it again, for sure I will include an user control to enable/disable, but maybe it will be enabled by default.

closed 2019-05-18 by RavenSystem

ivorocha90, 2019-05-18:
Sure. Thanks again.