RavenSystem/esp-homekit-devices #389: You have a italian hater

Francisco9696, 2019-05-15:
in italy there is a man who sells firmware like yours and makes lots of money and speaks badly of you in his group. 
maybe ruin his business... 
Look
[image]

GPL71, 2019-05-15:
@Francisco9696 
unfortunatly this happened last week, and not was a good advertising for italians
User and Homekit Italia Group have no relation with that person.
The post was against Homekit Italia Group too because support RavenSystem that work for free.
That post joined about 90 person to Homekit Italia Group in 2 days, and now they're are playing for fre with domotica and they're searching RavenSystem firmware.

This is the most important thing

greensouth, 2019-05-15:
Tontos hay en tos laos!

lizzus, 2019-05-15:
no comment!  the absurd thing is that that change to the firmware was requested here on github by a user who is part of his group!  Jose, don't consider what Mirko writes.  apparently as he writes @francisco9696 your work is making him lose money illegally!

RavenSystem, 2019-05-15:
This is hilarious. I totally do not know the work of that person and I will not value anything about their comments.

Just remember what the esp-homekit-devices license says in point 10: "Although already forbidden by the sources and subsequent licensing, it is not allowed to use or distribute this software for a commercial purpose."
https://github.com/RavenSystem/esp-homekit-devices/blob/master/LICENSE

**In addition to what Apple says, which is the most important:
https://developer.apple.com/support/homekit-accessory-protocol/**

That's why my firmwares are totally free. You can download, install and use them without having to make any payment or registration in any web or social group. And all the source code and the instructions to compile them are also available for free on this GitHub repo.

seritos, 2019-05-15:
We shouldnt allow this behaviour. We should all try to close these webs/pages. The work of ravencore must be respected and protected!!

GPL71, 2019-05-16:
@seritos I agree. 
Unfortunatly if user continue pay for firmware, someone continue selling.

In our facebook group and forum too all is very clear. 
Is very clear RavenCore is for free.
Translation of guides are for free and authorized by RavenSystem.
When people ask for price, we link RavenSystem donation.
Often they're surprised because they explain us that they pay for firmware.
Some user paid high costs... 
The users of our Group know the original links of firmware and each one of them can reach Raven here.
We encourage them partecipate them here in the issue, infact that is.

The right thing is ban from this community the person that use free work for commercial pourpose.

RavenSystem, 2019-05-18:
I think this topic can be closed. If anyone has more questions about, feel free to ask.

closed 2019-05-18 by RavenSystem