RavenSystem/esp-homekit-devices #390: problem to connect shelly 2.5

pilot1981, 2019-05-16:
Hi,
I successful used your firmware with sonoff basic and shelly: great job!

Now, I'm trying to use RavenCore 0.8.10 with shelly 2.5; I successful flashed device but I'm not able to connect it to apple homekit; connecting via console I receive only these messages:

aid 11
cnt 

connected with Vodafone-A76824481, channel 11
dhcp client start...
ip:192.168.1.127,mask:255.255.255.0,gw:192.168.1.1
scandone
scandone
scandone
scandone
scandone
scandone
scandone
scandone
scandone
scandone
scandone
scandone
scandone
scandone
scandone
scandone
scandone

It seem connected to my WIFI network, but why I can add it in homekit? home app don't find it...

P.S: I notice WIFI range of shelly 2.5 is less than shelly 1

RavenSystem, 2019-05-16:
Can you post the complete log?

pilot1981, 2019-05-17:
I have only this log...no other logs...you need log before or after WIFI connection?

pilot1981, 2019-05-17:
in previous attempt, I collected this log were shelly 2.5 wasn't able to connect to WIFI:

scandone
reconnect
scandone
add 0
aid 11
cnt

connected with Vodafone-A76824481, channel 11
dhcp client start...
>>> wifi_config: wifi_config_sta_connect_timeout_callback: Timeout connecting to WiFi network, starting config AP
>>> wifi_config: Starting AP mode
mode : sta(80:7d:3a:74:53:d9) + softAP(82:7d:3a:74:53:d9)
add if1
bcn 100
Function called without core lock
>>> wifi_config: wifi_config_softap_start: Starting AP SSID=LCM-7453D9
bcn 0
del if1
add if1
bcn 100
>>> wifi_config: Starting DHCP server
>>> wifi_config: Starting WiFi scan
>>> wifi_config: Starting DNS server
>>> wifi_config: Starting HTTP server
pp.c 603

it's sufficient?

RavenSystem, 2019-05-17:
In last log, I see that firmware running is LCM (OTA install) and not RavenCore. Maybe RavenCore is not fully installed because wifi lost in middle of installation process.

pilot1981, 2019-05-17:
Ok, so after WIFI config it need to be connected to WIFI for about 7 minutes? It could be related to WIFI problem of my Shelly 2.5...

RavenSystem, 2019-05-17:
7 minutes if your wifi works well. Of course, it is a problem with your Shelly2.5 wifi.

pilot1981, 2019-05-17:
I tried again...erased again...flashed again...configured WIFI near WIFI router: but I have always the same result: no accessory added in apple home app; this is log from console:

aid 9
cnt 
scandone
reconnect
rm match
scandone
add 0
aid 9
cnt 
scandone
rm match
reconnect
scandone
add 0
aid 9
cnt 
scandone
reconnect
rm match
scandone
add 0
aid 9
cnt 

what mean? always WIFI problems?

pilot1981, 2019-05-17:
I tried again and again and now shelly 2.5 is paired with apple home app, but I think this is something wrong, please see this part of logs:

mDNS announcement: Name=RavenCore-7453D9 md=RavenCorepv=1.0id=E3:2F:36:F2:E6:34c#=522s#=1ff=0sf=1ci=8sh=WfUsFA== Port=5556 TTL=4500
>>> HomeKit: Got new client connection: 4
>>> HomeKit: [Client 4] Pair Setup Step 1/3
>>> HomeKit: [Client 4] Pair Setup Step 2/3
!!! HomeKit: [Client 4] Failed to compute SRP shared secret (code -2)
>>> HomeKit: [Client 4] Pair Setup Step 1/3
>>> HomeKit: [Client 4] Pair Setup Step 2/3
!!! HomeKit: [Client 4] Failed to compute SRP shared secret (code -2)
>>> HomeKit: [Client 4] Pair Setup Step 1/3
>>> HomeKit: [Client 4] Closing client connection
>>> HomeKit: Got new client connection: 4
>>> HomeKit: [Client 4] Pair Setup Step 1/3
>>> HomeKit: [Client 4] Pair Setup Step 2/3
>>> mdns_recv, could not alloc 1001
>>> mdns_reply could not alloc 1460
>>> HomeKit: [Client 4] Pair Setup Step 3/3
>>> HomeKit: Added pairing with 226B3592-2307-44C6-9C4D-E087492C6239
>>> HomeKit: Configuring mDNS
mDNS announcement: Name=RavenCore-7453D9 md=RavenCorepv=1.0id=E3:2F:36:F2:E6:34c#=522s#=1ff=0sf=0ci=8sh=WfUsFA== Port=5556 TTL=4500
>>> HomeKit: [Client 4] Successfully paired
>>> HomeKit: Got new client connection: 5
>>> HomeKit: [Client 4] Closing client connection
>>> HomeKit: [Client 5] Pair Verify Step 1/2
>>> HomeKit: [Client 5] Pair Verify Step 2/2
>>> HomeKit: [Client 5] Found pairing with 226B3592-2307-44C6-9C4D-E087492C6239
>>> HomeKit: [Client 5] Verification successful, secure session established
>>> HomeKit: [Client 5] Get Accessories
>>> HomeKit: [Client 5] Add Pairing
>>> HomeKit: Added pairing with BDD9D4D4-7949-4C18-9505-1222EF2FA05C
>>> HomeKit: [Client 5] Update Characteristics
>>> HomeKit: [Client 5] Update Characteristics
>>> HomeKit: [Client 5] Get Characteristics
>>> HomeKit: [Client 5] Update Characteristics
RC > Identifying
>>> HomeKit: Got new client connection: 4
>>> HomeKit: [Client 4] Pair Verify Step 1/2
>>> HomeKit: [Client 4] Pair Verify Step 2/2
>>> HomeKit: [Client 4] Found pairing with 226B3592-2307-44C6-9C4D-E087492C6239
>>> HomeKit: [Client 4] Verification successful, secure session established
>>> HomeKit: [Client 4] Get Accessories
>>> HomeKit: Got new client connection: 6
>>> HomeKit: [Client 6] Pair Verify Step 1/2
>>> HomeKit: [Client 6] Pair Verify Step 2/2
>>> HomeKit: [Client 6] Found pairing with 226B3592-2307-44C6-9C4D-E087492C6239
>>> HomeKit: [Client 6] Verification successful, secure session established
>>> HomeKit: [Client 6] Get Accessories

Im near router WIFI, with other devices shelly 1 and sonoff I never had all this type of error during pairing procedure....what do you think?

pilot1981, 2019-05-17:
the WIFI problem remain: if go away from router WIFI I lost device from apple home app...the same result with another shelly 2.5...I don't understand...

Another question: it's possibile use 1 shelly 2.5 and move 1 relais to room A and 1 relais in room B of apple home app? or using shelly 2.5 I must have both relais in the same room?

RavenSystem, 2019-05-17:
Some Shellys 2.5 have problem with antenna. Check it: https://www.facebook.com/groups/ShellyIoTCommunitySupport/permalink/2029876037111750/?__tn__=CH-R

> Another question: it's possibile use 1 shelly 2.5 and move 1 relais to room A and 1 relais in room B of apple home app?

No with RC. Yes with HAA.

pilot1981, 2019-05-18:
My Shelly 2.5 don’t have problem of WIFI ANTENNA/PIN...no hole...

pilot1981, 2019-05-18:
@RavenSystem what you say about problems of find new device and problems of pairing?

pilot1981, 2019-05-20:
Problem seems solved with firmware 1.4.9 of Shelly.cloud...it’s possible? Now how I can solve with Shelly 2.5 already flashed to ravencore firmware?

pilot1981, 2019-05-20:
Shelly support say to me there is a problem on SDK of ESP and they do a workaround on firmware 1.5.0 of Shelly.cloud (it will be released tomorrow)

RavenSystem, 2019-05-20:
That is very strange. Many people uses Shelly2.5 without any problem. And I don't use SDK of ESP, I use esp-open-rtos.

pilot1981, 2019-05-21:
yes, It's very strange also for me...I think there are different version of shelly 2.5 (as for shelly 1), and some shelly 2.5 could have this type of problem..now I don't understand how can I have the same WIFI problem on shelly 2.5 using shelly firmware 20190319-100741/master and your firmware...then shelly support solve the problem using 1.4.9 version and today shelly support release 1.5.0 version to fix better this problem...it's all very strange...I don't understand where is the problem...what is the truth?

RavenSystem, 2019-05-23:
What version of Shelly 2.5 have you?

pilot1981, 2019-05-23:
firmware 20190319-100741/master

RavenSystem, 2019-05-23:
I was asking about hardware version/revision.

pilot1981, 2019-05-24:
I don't know! where I can see it?

RavenSystem, 2019-05-26:
Sorry, I don't know. Actually I have not got any Shelly2.5.

pilot1981, 2019-05-27:
I think this could be an antenna problem...I'm waiting for new WIFI antenna from shelly.cloud company...

DanielMosco, 2019-05-28:
Hi guys. I have successfully flashed sonoff devices with ravencore using the instructions that are available all over the web. How i flash the shelly 2.5? How to put it into pairing mode and is the rest of the process is similar to the sonoff one. Any help is appreciated. Thanks in advance.

i3laze, 2019-06-08:
As pins on Shelly 2.5 are exposed, holes are smaller and standard breadboard cables don’t fit.
I still had to open the case up and use needles to flash it.

Flash Mode initiated by connecting to GPIO0 to GND before powering it up:
https://blakadder.github.io/templates/shelly_25.html

i3laze, 2019-06-08:
As pins on Shelly 2.5 are exposed, holes are smaller and standard breadboard cables don’t fit.
I still had to open the case up and use needles to flash it.
[image]

Flash Mode initiated by connecting GPIO0 to GND.

DanielMosco, 2019-06-08:
Thank you very much!!!

pilot1981, 2019-06-19:
Problem solved changing WIFI ANTENNA

closed 2019-06-19 by pilot1981