RavenSystem/esp-homekit-devices #391: new device with8.10 cannot pair

douzs, 2019-05-17:
Hi, I really love your firmwares,they are so great,and thank you for creating those.
But I found that there seems to be something wrong.
When I removed the device in my homekit, there was no way to add it again,the log tells it has been paird,and it won't be showed at homekit.
Another problem is that I use your latest version 8.10 to finish the firmware. When pairing, I will be prompted that the pin input is incorrect.
I hope that you have time to verify, I hope it is my problem.

RavenSystem, 2019-05-17:
When you remove an accessory from Home App, accessory must be connected and working to receive the reset command and then you can add it again to HomeKit; if not, you must do a factory reset in the accessory.

The other problem with pin pairing incorrect can be a problem with your mDNS environment. Try to reset your router wifi, your iOS device used to pair and your accessory, and try again.

douzs, 2019-05-17:
@RavenSystem thank you ,i will try

closed 2019-05-17 by douzs

RavenSystem, 2019-05-17:
Feel free to reopen issue if problem is not solved.