RavenSystem/esp-homekit-devices #392: Trying to test HAA 0.0.2

fluppie, 2019-05-17:
Sonoff classic, one off the earliest versions but already with GPIO14 on the pin header.

Getting this after erase flash, flash firmware and trying to connect to the HAA AP:

> scandone
> add 1
> aid 1
> station: 18:81:0e:5a:86:e7 join, AID = 1
> >>> wifi_config: http_task: Got 64 incomming data
> !!! wifi_config: Unknown endpoint: GET /hotspot-detect.html
> >>> wifi_config: http_task: Got 64 incomming data
> >>> wifi_config: http_task: Got 4 incomming data
> >>> wifi_config: wifi_config_server_on_message_complete: Unknown endpoint
> >>> wifi_config: client_send_redirect: Redirecting to http://192.168.4.1/settings
> >>> wifi_config: Client disconnected
> >>> wifi_config: http_task: Got 64 incomming data
> >>> wifi_config: http_task: Got 51 incomming data
> Fatal exception (28): 
> epc1=0x4000bf80
> epc2=0x00000000
> epc3=0x4020361c
> excvaddr=0x00000000
> depc=0x00000000
> excsave1=0x4021ef08
> Registers:
> a0 4021ef08 a1 3fff5dd0 a2  00000000 a3  fffffffc
> a4  000000ff a5  0000ff00 a6  00ff0000 a7  ff000000
> a8  00000100 a9  0000001a a10 00000018 a11 04000102
> a12 3fff1d88 a13 3fff5f33 SAR 00000001
> 
> Stack: SP=0x3fff5dd0
> 0x3fff5dd0: 0d636536 3fff000a 3fff1d88 40220b50
> 0x3fff5de0: 00010000 3fff5f33 3fff1d88 4021f070
> 0x3fff5df0: 4023cb73 00000000 3fff5ffc 40210b50
> 0x3fff5e00: 4023cb73 00000000 3fff5ffc 40106856
> 0x3fff5e10: 00000000 ffffffff ffffffff 00000000
> 0x3fff5e20: 00000002 00000032 00302064 40221988
> 0x3fff5e30: 00000040 00000000 00000033 40220398
> 0x3fff5e40: 00000000 00000000 3ffef5e0 40204ce2
> 
> Free Heap: 36012
> _heap_start 0x3fff1d00 brk 0x3fff7f50 supervisor sp 0x40000000 sp-brk 32944 bytes
> arena (total_size) 25168 fordblks (free_size) 3068 uordblocks (used_size) 22100
> 
>  ets Jan  8 2013,rst cause:1, boot mode:(3,7)
> 
> load 0x40100000, len 2292, room 16 
> tail 4
> chksum 0x57
> load 0x3ffe8000, len 772, room 4 
> tail 0
> chksum 0x0b
> csum 0x0b
> 
> rBoot v1.4.0 - richardaburton@gmail.com
> Flash Size:   8 Mbit
> Flash Mode:   DOUT
> Flash Speed:  40 MHz
> rBoot Option: Big flash
> rBoot Option: RTC data
> 
> Booting rom 0.
> pp_task_hdl : 3ffeff00, prio:14, stack:512
> pm_task_hdl : 3ffef7f0, prio:1, stack:176
> frc2_timer_task_hdl:0x3fff3e80, prio:12, stack:200
> 
> ESP-Open-SDK ver: 0.0.1 compiled @ May 17 2019 11:04:13
> phy ver: 273, pp ver: 8.3
> 
> HAA ! ERROR: Invalid JSON
[... 562 lines cut ...]
> csum 0x0b
> 
> rBoot v1.4.0 - richardaburton@gmail.com
> Flash Size:   8 Mbit
> Flash Mode:   DOUT
> Flash Speed:  40 MHz
> rBoot Option: Big flash
> rBoot Option: RTC data
> 
> Booting rom 0.
> pp_task_hdl : 3ffeff00, prio:14, stack:512
> pm_task_hdl : 3ffef7f0, prio:1, stack:176
> frc2_timer_task_hdl:0x3fff3e80, prio:12, stack:200
> 
> ESP-Open-SDK ver: 0.0.1 compiled @ May 17 2019 11:04:13
> phy ver: 273, pp ver: 8.3
> 
> HAA ! ERROR: Invalid JSON
> mode : sta(5c:cf:7f:1b:d8:9b) + softAP(5e:cf:7f:1b:d8:9b)
> add if0
> add if1
> bcn 100
> add 1
> aid 1
> station: 18:81:0e:5a:86:e7 join, AID = 1

GuyCoast, 2019-05-17:
same problem with esp01

RavenSystem, 2019-05-17:
I know.
v0.0.2 has been retired.
v0.0.3 is on the way.

GuyCoast, 2019-05-17:
> I know.
> v0.0.2 has been retired.
> v0.0.3 is on the way.
> 
> Betas are betas.

Many thanks Raven

RavenSystem, 2019-05-18:
Fixed in v0.0.3. It is released. Many thanks for your feedbacks.

closed 2019-05-18 by RavenSystem

fluppie, 2019-05-18:
Yes, working now :)
```

rBoot v1.4.0 - richardaburton@gmail.com
Flash Size:   8 Mbit
Flash Mode:   DOUT
Flash Speed:  40 MHz
rBoot Option: Big flash
rBoot Option: RTC data

Booting rom 0.
pp_task_hdl : 3ffeff10, prio:14, stack:512
pm_task_hdl : 3ffef800, prio:1, stack:176
frc2_timer_task_hdl:0x3fff3ea8, prio:12, stack:200

ESP-Open-SDK ver: 0.0.1 compiled @ May 18 2019 12:32:14
phy ver: 273, pp ver: 8.3

HAA > Home Accessory Architect
HAA > Developed by @RavenSystem - José Antonio Jiménez
HAA > Version: 0.0.3

HAA > Running in NORMAL mode...
HAA > ----- PROCESSING JSON -----
HAA > Enable UART log output
HAA > Enable LED GPIO 13, inverted=1
HAA > Creating Acc 0, type=1
HAA > Enable button GPIO 0, type=1, inverted=0
HAA > Enable button GPIO 14, type=0, inverted=0
HAA > Enable button GPIO 14, type=1, inverted=0
HAA > Enable output GPIO 12
HAA > Output GPIO 12 -> 0
HAA > Free Heap: 45976
HAA > ---------------------------
>>> wifi_config: Initializing WiFi config
>>> wifi_config: Found configuration, connecting to
```