RavenSystem/esp-homekit-devices #393: Hide switch on device type 6

mafyata, 2019-05-17:
Hi , I have many devices with your software. It is wonderfull. Can you make on/off switch(eve app setup) to hide switch on device type 6. I have NodeMcu and attached DHT22 sensor to it. I flashed it with your firmware and change it to device type 6 to monitor temp and humidity, but I see the switch button. Can we hide this switch button because it is unused, or at device type 16 to make a sensor for temp and humidity?

greensouth, 2019-05-17:
That is exactly my plan!

I’ve ordered some nodeMCU to attach a motor and build a watering system. I suppose that Jose will add support for more types of devices.

mafyata, 2019-05-17:
[image]

RavenSystem, 2019-05-23:
RavenCore will not support that. HAA, yes.

closed 2019-05-26 by RavenSystem