RavenSystem/esp-homekit-devices #394: HAA Relay status LED

dwhp, 2019-05-18:
Hi. 
Love HAA. I have some random amazon purchased WiFi sockets, programmed with haa successfully. Connected to the gpio’s it has: 
Gpio 12 = Relay
Gpio 13 = momentary  Switch, 
Gpio 04 = blue LED and 
Gpio 05 = RED LED. 

To add the relay red on/off LED I added it to the relay array and inverted it as LOW=ON

Working with the exception of first boot, where gpio 5 is low (red LED on). AND gpio 12 is low (Relay as expected. Turning on off puts it all back in sink. 

Question is there a better way I missed/planned for the red relay status led OR if an output is inverted, can it be inverted on boot? 

My code:
{"c":{"o":1,"l":4,"i":1},"a":[{"t":1,"r":[{"g":12},{"g":5,"i":1}],"b":[{"g":13,"p":1,"t":1}]}]}

dwhp, 2019-05-18:
Hope above was clear.

Needs to be when off (works):
Gpio 12 LOW (Relay)
Gpio 5 HIGH (RED LED OFF)

Needs to be when on (works): 
Gpio 12 HIGH (Relay)
Gpio 5 LOW (RED LED ON)

When first boots (issue = red led on & Relay off):
Gpio 12 LOW (Relay)
Gpio 5 LOW (RED LED ON - needs to be high = off)

Switching Relay on and off again either via app or momentary switch, puts the LED in sync with intention. It’s only wrong after first boot. Ie socket off but led is on.

RavenSystem, 2019-05-18:
@dwhp Your setup is right. Problem is actual version doesn't support initial inverted status at boot with relays. I am going to release next version soon and it will have it.

RavenSystem, 2019-05-18:
Fixed in v0.0.3. It is released.

closed 2019-05-18 by RavenSystem

dwhp, 2019-05-19:
Just updated. Works like a charm. Thanks :)