RavenSystem/esp-homekit-devices #395: Can't connect to HAA Web GUI

gijsstegehuis, 2019-05-18:
Device is running in setup mode (see logs below), but i can't connect to the web GUI at 192.168.0.112. Is it running on a different port, or am i missing something else?

```Booting rom 0.
pp_task_hdl : 3ffeff90, prio:14, stack:512
pm_task_hdl : 3ffef880, prio:1, stack:176
frc2_timer_task_hdl:0x3fff3f20, prio:12, stack:200

ESP-Open-SDK ver: 0.0.1 compiled @ May 18 2019 11:46:26
phy ver: 273, pp ver: 8.3

HAA > Running in SETUP mode...
>>> wifi_config: Initializing WiFi config
>>> wifi_config: Found configuration, connecting to prettyflyforawifi
mode : sta(cc:50:e3:25:a8:83)
add if0
scandone
add 0
aid 3
cnt

connected with prettyflyforawifi, channel 11
dhcp client start...
ip:192.168.0.112,mask:255.255.255.0,gw:192.168.0.1
>>> wifi_config: wifi_config_sta_connect_timeout_callback: Successfully connected```

RavenSystem, 2019-05-18:
You are doing all right. Are you using last version of HAA (0.0.3)??

RavenSystem, 2019-05-18:
Please, try version 0.0.4 and comment. Thanks!

gijsstegehuis, 2019-05-19:
@RavenSystem correct, i was using 0.0.3.
Just tested it on 0.0.4. It's working fine now. Thanks 👍

closed 2019-05-19 by gijsstegehuis