RavenSystem/esp-homekit-devices #396: HAA configuring accessory as stateless switch

gijsstegehuis, 2019-05-19:
I configured my button to support single, double and long press, but they show up as separate Homekit buttons that way. Can i configure them to show up as a single stateless Homekit button?

This is the configuration i am using at the moment:
```{
  "a": [
    {
      "r": [
        {
          "g": 4
        }
      ],
      "b": [
        {
          "g": 5,
          "p": 0
        }
      ]
    },
    {
      "r": [
        {
          "g": 4
        }
      ],
      "b": [
        {
          "g": 5,
          "t": 2
        }
      ]
    },
    {
      "r": [
        {
          "g": 4
        }
      ],
      "b": [
        {
          "g": 5,
          "t": 3
        }
      ]
    }
  ]
}```

RavenSystem, 2019-05-19:
Stateless Homekit button is not supported yet.

RavenSystem, 2019-05-26:
Supported: https://github.com/RavenSystem/esp-homekit-devices/releases/tag/HAA_0.0.5

closed 2019-05-26 by RavenSystem