RavenSystem/esp-homekit-devices #397: Problem with pairing when board 4 was choosen

kompak23, 2019-05-19:
Hi,
I am very satisfied with your software, but I have one problem. 
When I change board type: 4 and then reset Sonoff to default settings - it is unable to pair with Homekit.
Before changing - everithing work well. 
I need this board type to connect to RX - DS18B20 or DHT sensor.

Have you an idea why it is like so? 

Thanks for reply.

RavenSystem, 2019-05-20:
Have you try to repower device?

kompak23, 2019-05-20:
Yes, many times. 
But problem is only when I choose board type: 4.

RavenSystem, 2019-05-20:
Can you pair with other type board selected and then change to 4 once it is paired??

kompak23, 2019-05-20:
Flashed device which has default setted "device type = 1" works well. 
After pairing, when I change device type to 4 and then I want to reset every value to default  by pushing button 10 second, then It is unable to pair with HomeApp.

RavenSystem, 2019-05-20:
But why do you want to do a full reset?

kompak23, 2019-05-20:
Maybe it is enough to do Wifi Reset, but after should I do pairing again?

RavenSystem, 2019-05-20:
Wifi reset will not remove pairing info.

RavenSystem, 2019-06-13:
Fixed in v0.9.3.

closed 2019-06-13 by RavenSystem