RavenSystem/esp-homekit-devices #398: question double buttons

curla92, 2019-05-19:
Hi josè, i would to know if possible clone 1 button with firmware haa. i have a lamp with 2 button in kitchen and living room, i want 2 button replicated in homekit. thanks

RavenSystem, 2019-05-19:
I am not sure what you want...

curla92, 2019-05-19:
Ok. I have a lamp with two physical buttons(1 in kitchen and 1 in living room). I would to replicated these same buttons in homekit with a shelly. I know that it’s impossible divided 2 toggles in 2 different rooms in homekit. I hope it’s clear. Thanks

RavenSystem, 2019-05-20:
HAA supports multiroom. You can place each switch in different Homekit rooms. But you will need automations to do what you want.

curla92, 2019-05-20:
resolved. it's work perfectly. very thanks josè

closed 2019-05-20 by RavenSystem