RavenSystem/esp-homekit-devices #399: Water valve lost wifi BUG

netRunner0, 2019-05-20:
Hey. I noticed such a feature that when Sonoff 12v loses the network or reconnects to another point, it returns the relay to its original position and closes the water for me)

mafyata, 2019-05-20:
Did you set correct Init State SW via eve app ?

netRunner0, 2019-05-20:
Init State - 2 0 not working ]

RavenSystem, 2019-05-20:
@netRunner0 What is the bug? I don't understand if you want a feature that closes relay when wifi is lost...

netRunner0, 2019-05-24:
Yes, I would like the relay to not change its position when the network is lost)

RavenSystem, 2019-05-24:
Relay doesn’t change when wifi is lost.

netRunner0, 2019-06-18:
https://www.itead.cc/ru/inching-self-locking-wifi-wireless-switch.html

I used normal and reverse connections, but once every 1-2 weeks the device either reboots or loses the network and changes the position of the relay.

i3laze, 2019-08-08:
What exactly happens when you power cycle?
Did you try different Initial State settings?

closed 2019-11-07 by RavenSystem