RavenSystem/esp-homekit-devices #404: Thermostat + Si7021

skanico, 2019-05-29:
Hello,
After some research, I found some topics explaining the protocol used by Si7021.

[https://github.com/arendst/Sonoff-Tasmota/issues/735#issuecomment-348718383](url)

and some source code :

[https://github.com/arendst/Sonoff-Tasmota/commit/253d60fd2b0c3d452d0dd8798ff767a1b143be98](url)

Maybe this could help you add support for this probe.

Thanks

mafyata, 2019-05-29:
It will be very good if device type 6 support Si7021 sensor. I have sonoff th16 with Si7021 sensor and want to use it. Sonoff recommend to use this sensor  with th10/th16.

skanico, 2019-05-30:
I had a look in the code, and it seems the implementation of the Si7021 has to be done in esp-open-rtos. I have opened an issue there.

RavenSystem, 2019-05-30:
If esp-open-rtos supports Si7021 finally, for sure that I will add it in RC and HAA.

mafyata, 2019-06-07:
I have th16 + si7021 and can test it if you work on it.

mafyata, 2019-06-07:
If it help:
https://github.com/arendst/Sonoff-Tasmota/commit/253d60fd2b0c3d452d0dd8798ff767a1b143be98

RavenSystem, 2019-07-20:
https://github.com/SuperHouse/esp-open-rtos/issues/719

closed 2019-07-20 by RavenSystem

oremic, 2019-10-09:
Hi my th10 and Itead si7021 not work
[image]

mafyata, 2019-10-09:
Push the plug harder to th10

oremic, 2019-10-09:
> Push the plug harder to th10
read value no real.
TEMP 0, HUM 2
Si7021: 2
dht: Sensor data: humidity=20, temp=0

mafyata, 2019-10-09:
Device type: 6
TH Sensor Type: 4
I have th16 + Si7021 and work.

oremic, 2019-10-09:
Same my photo sensor

ateodorescu, 2019-10-15:
First of all congrats for the great work you did with this firmware. I have a bunch of Sonoffs and Shellys that already work with Ravencore and wanted to have temp and humidity sensors too.
I have both th10 and th16 (v2.1 for both of them) and a Si7021 sensor. They both don't work.
I have configured in Eve boardType 1, deviceType 6, sensorType 4 and I only get something like temp = 1 and humidity 1. Not sure what's wrong. Any way I can debug this?

Thanks!

xrust83, 2019-12-02:
temp = 1 and humidity 1
I have the same result