RavenSystem/esp-homekit-devices #405: Dimmer Light

walvespit, 2019-05-30:
Hello, Thank you for this amazing project!!

It is possible with any configuration to dimmer a lamp?

mafyata, 2019-05-30:
You want this ?
[image]

walvespit, 2019-05-30:
> You want this ?
> [image]

Yes, Exactly!

mafyata, 2019-05-30:
This is rgb led strip. I don’t know light bulb with dimmer settings at ravencore. Waiting the Boss to reply :)

RavenSystem, 2019-05-30:
> It is possible with any configuration to dimmer a lamp?

Actually, no. And RC will not support it, and anything new. But Home Accessory Architect will do it.

walvespit, 2019-05-30:
> > It is possible with any configuration to dimmer a lamp?
> 
> Actually, no. And RC will not support it, and anything new. But Home Accessory Architect will do it.

Thank you! By the way, i was just checking HAA now and i think at this time RavenCore have more options right? For example i use a sonoff with device type 13, it is not possible in HAA yet, thats correct?

RavenSystem, 2019-05-30:
Yes. RC is near of its end of roadmap, while HAA is in a early stage. But actually HAA can do many things that RC can not with switches, outlets and stateless buttons.

1nutty1, 2019-07-01:
Would the first of your HAA Templates be the correct one to use with a dimmer that has 3 touch switches? (on/off, brighter and dimmer)
Such as the following:
https://github.com/arendst/Sonoff-Tasmota/wiki/Dimmer-switch-with-Nuvoton-chip

RavenSystem, 2019-07-01:
Dimmer is not supported yet.

1nutty1, 2019-07-01:
Thanks for the swift response.
Any idea as to when this might be looked at?

RavenSystem, 2019-07-20:
> Any idea as to when this might be looked at?

No, sorry.

closed 2019-07-20 by RavenSystem