RavenSystem/esp-homekit-devices #406: raven code

mohamedkeven, 2019-05-31:
hi i installed ravencore v9 main.bin with blank and rboot bin but when i add in homekit i have to enter accessory code but i haven't i test 11111111 but not work can you giveme the code

mafyata, 2019-05-31:
Read wiki.
[image]

RavenSystem, 2019-05-31:
https://github.com/RavenSystem/esp-homekit-devices/wiki/RavenCore

vojtechtambor, 2019-06-07:
just flashed v 0.90 onto a sonoff basic. wifi connected OK, device discoverable in HomeKit

pairing refuses both "02182017" as well as "11111111", tried QR, tried manually entering, tried power cycling several times.. 

Sonoff unit works using manual button to toggle the lamp

thanks

RavenSystem, 2019-06-10:
Code was, is and will be 02182017 forever and ever. Try version 0.9.1.

closed 2019-06-12 by RavenSystem