RavenSystem/esp-homekit-devices #409: Type 12 Window Covering doesn’t save state

i3laze, 2019-06-02:
Subject.
I’m using Sonoff Dual R2. When I shut down the power, the device doesn’t save last state and shows Closed on startup. 
Of course there are no physical sensors to detect actual state, but let’s assume Sonoff is the only device controlling the window.

Issue 2: Button on Dual R2 doesn’t work:
Although Button worked for Device Type 2 (Dual Switch, triggers SW1), it does nothing for Type 12:
!!! AdvButton: No function defined
I've tried all available Board Types.

RavenSystem, 2019-06-09:
Built-in button has not any function to open, close or stop.

i3laze, 2019-06-09:
Please, make it to toggle the state.

RavenSystem, 2019-06-10:
Try version 0.9.1.

i3laze, 2019-06-14:
Sorry for a delayed feedback.

Regardless board type 1,2,4 button Opens.
Despite my original request was state toggle.. open will solve my current task :)

RavenSystem, 2019-06-14:
And double press close.

closed 2019-06-14 by RavenSystem