RavenSystem/esp-homekit-devices #410: Pairing issues v0.9.0

fluppie, 2019-06-03:
Any idea why the old pairing has issues in 0.9.0?
```

>>> HomeKit: Got new client connection: 4
>>> HomeKit: [Client 4] Pair Setup Step 1/3
>>> HomeKit: [Client 4] Pair Setup Step 2/3
!!! HomeKit: [Client 4] Failed to compute SRP shared secret (code -2)
>>> HomeKit: [Client 4] Pair Setup Step 1/3
>>> HomeKit: [Client 4] Pair Setup Step 2/3
!!! HomeKit: [Client 4] Failed to compute SRP shared secret (code -2)
>>> HomeKit: [Client 4] Pair Setup Step 1/3
>>> HomeKit: [Client 4] Pair Setup Step 2/3
>>> mdns_reply could not alloc 1460
>>> mdns_reply could not alloc 1460
!!! HomeKit: [Client 4] Failed to verify peer's proof (code -219)
>>> HomeKit: [Client 4] Pair Setup Step 1/3
```

```
mDNS announcement: Name=RavenCore-1BD89B md=RavenCorepv=1.0id=AF:0D:54:52:15:91c#=576s#=1ff=0sf=1ci=8sh=9uoqwA== Port=5556 TTL=4500
>>> HomeKit: Got new client connection: 4
>>> HomeKit: [Client 4] Pair Setup Step 1/3
>>> HomeKit: [Client 4] Pair Setup Step 2/3
!!! HomeKit: [Client 4] Failed to compute SRP shared secret (code -2)

```

Here I tried the QR code I found in #371 

```

>>> HomeKit: [Client 4] Pair Setup Step 1/3
>>> HomeKit: [Client 4] Closing client connection
>>> HomeKit: Got new client connection: 4
>>> HomeKit: [Client 4] Pair Setup Step 1/3
>>> HomeKit: [Client 4] Pair Setup Step 2/3
>>> HomeKit: [Client 4] Pair Setup Step 3/3
>>> HomeKit: Added pairing with E6A69006-2669-4B56-84A7-17776349F804
>>> HomeKit: Configuring mDNS
mDNS announcement: Name=RavenCore-1BD89B md=RavenCorepv=1.0id=AF:0D:54:52:15:91c#=576s#=1ff=0sf=0ci=8sh=9uoqwA== Port=5556 TTL=4500
>>> HomeKit: [Client 4] Successfully paired
>>> HomeKit: Got new client connection: 5
>>> HomeKit: [Client 4] Closing client connection
>>> HomeKit: [Client 5] Pair Verify Step 1/2
>>> HomeKit: [Client 5] Pair Verify Step 2/2
>>> HomeKit: [Client 5] Found pairing with E6A69006-2669-4B56-84A7-17776349F804
>>> HomeKit: [Client 5] Verification successful, secure session established
>>> HomeKit: [Client 5] Get Accessories
>>> HomeKit: [Client 5] Add Pairing
>>> HomeKit: Added pairing with ECAD2707-EB17-4DC1-98E5-5FFD9445C29C
>>> HomeKit: [Client 5] Add Pairing
>>> HomeKit: Added pairing with 26CCB35C-DE2F-4BA0-BBCD-076C9C759200
>>> HomeKit: [Client 5] Add Pairing
>>> HomeKit: Added pairing with 46883629-01D7-4ACA-B4F8-218040199EA1
>>> HomeKit: [Client 5] Update Characteristics
>>> HomeKit: [Client 5] Update Characteristics
>>> HomeKit: [Client 5] Get Characteristics
>>> HomeKit: Got new client connection: 4
>>> HomeKit: [Client 4] Pair Verify Step 1/2
>>> HomeKit: [Client 5] Update Characteristics
RC > Identifying
>>> HomeKit: [Client 4] Pair Verify Step 2/2
>>> HomeKit: [Client 4] Found pairing with E6A69006-2669-4B56-84A7-17776349F804
>>> HomeKit: [Client 4] Verification successful, secure session established
>>> HomeKit: [Client 4] Get Accessories
>>> HomeKit: Got new client connection: 6
>>> HomeKit: [Client 6] Pair Verify Step 1/2
>>> HomeKit: [Client 6] Pair Verify Step 2/2
>>> HomeKit: [Client 6] Found pairing with E6A69006-2669-4B56-84A7-17776349F804
>>> HomeKit: [Client 6] Verification successful, secure session established
>>> HomeKit: [Client 6] Get Accessories
>>> HomeKit: Got new client connection: 7
>>> HomeKit: [Client 7] Pair Verify Step 1/2
>>> HomeKit: Got new client connection: 8
>>> HomeKit: [Client 7] Pair Verify Step 2/2
>>> HomeKit: [Client 7] Found pairing with E6A69006-2669-4B56-84A7-17776349F804
>>> HomeKit: [Client 7] Verification successful, secure session established
>>> HomeKit: [Client 8] Pair Verify Step 1/2
>>> HomeKit: [Client 7] Get Accessories
>>> HomeKit: [Client 8] Pair Verify Step 2/2
>>> HomeKit: [Client 8] Found pairing with E6A69006-2669-4B56-84A7-17776349F804
>>> HomeKit: [Client 8] Verification successful, secure session established
>>> HomeKit: [Client 8] Get Accessories
>>> HomeKit: [Client 4] Update Characteristics
>>> HomeKit: [Client 5] Get Characteristics
>>> HomeKit: [Client 5] Update Characteristics
RC > Toggle SW 1
RC > Relay 1 -> 1
>>> HomeKit: [Client 5] Update Characteristics
RC > Toggle SW 1
RC > Relay 1 -> 0
RC > Saving last states
```
Or pure luck that it worked through the QR code?

Partizan7676, 2019-06-08:
something strange. device name contain last symbols from its mac - in your logs they are different.
Name=RavenCore-**1BD89B** md=RavenCore�pv=1.0�id=AF:0D:54:**52:15:91**

VladNord, 2019-06-08:
I have problem with connecting ravencore device to wi-fi.
After entering password for wi-fi and OTA repository nothing happens.
ESP Device still in in AP mode

RavenSystem, 2019-06-12:
Try pairing with version 0.9.2.

kiralikbeyin, 2019-06-13:
I have th10 

uint8_t device_type_static = 5

````
>>> HomeKit: [Client 4] Pair Setup Step 1/3
>>> HomeKit: [Client 4] Pair Setup Step 2/3
!!! HomeKit: [Client 4] Failed to compute SRP shared secret (code -2)
>>> HomeKit: [Client 4] Pair Setup Step 1/3
````
latest 0.9.2

I see it like normal switch in homekit setup

mafyata, 2019-06-13:
Did you have installed before in homekit ?

kiralikbeyin, 2019-06-13:
yes it was working in 0.9.1

kiralikbeyin, 2019-06-13:
@VladNord  if your device is th10 dont plug sensors until you connect your wifi router

RavenSystem, 2019-06-13:
@kiralikbeyin I have tested a clean install of last version (0.9.3) just now and wifi setup and pairing process work perfectly. And I use some TH10 as heater thermostats and all of them works fine. Can you post a screenshot of your config device?

mafyata, 2019-06-13:
You must delete old device setup in homekit and then add it again in homekit.

kiralikbeyin, 2019-06-13:
@RavenSystem 

I also installed 0.9.3

````
>>> HomeKit: Configuring mDNS
RC > Saving last states
mDNS announcement: Name=RavenCore-88EA9C 
                                         md=RavenCorepv=1.0id=FC:E1:AC:2E:94:ACc#=579s#=1ff=0sf=1ci=8
                                                                                                     sh=KRSfoA== Port=5556 TTL=4500
>>> HomeKit: Got new client connection: 4
>>> HomeKit: [Client 4] Pair Setup Step 1/3
>>> HomeKit: [Client 4] Pair Setup Step 2/3
>>> mdns_reply could not alloc 1460
>>> HomeKit: [Client 4] Pair Setup Step 3/3
>>> HomeKit: Added pairing with 351A3A33-91F2-4949-B527-94FE963EBC0A
>>> HomeKit: Configuring mDNS
mDNS announcement: Name=RavenCore-88EA9C 
                                         md=RavenCorepv=1.0id=FC:E1:AC:2E:94:ACc#=579s#=1ff=0sf=0ci=8
                                                                                                     sh=KRSfoA== Port=5556 TTL=4500
>>> HomeKit: [Client 4] Successfully paired
>>> HomeKit: Got new client connection: 5
>>> HomeKit: [Client 4] Closing client connection
>>> HomeKit: [Client 5] Pair Verify Step 1/2
>>> HomeKit: [Client 5] Pair Verify Step 2/2
>>> HomeKit: [Client 5] Found pairing with 351A3A33-91F2-4949-B527-94FE963EBC0A
>>> HomeKit: [Client 5] Verification successful, secure session established
>>> HomeKit: [Client 5] Get Accessories
>>> HomeKit: [Client 5] Get Characteristics
>>> HomeKit: [Client 5] Update Characteristics
>>> HomeKit: [Client 5] Update Characteristics
>>> HomeKit: [Client 5] Add Pairing
>>> HomeKit: Added pairing with 57053BA1-2B2D-444E-B8EE-400A9B688977
>>> HomeKit: [Client 5] Get Characteristics
>>> HomeKit: [Client 5] Add Pairing
>>> HomeKit: Added pairing with AA02D5FE-7A85-44AD-943B-160FD404203E
>>> HomeKit: [Client 5] Get Characteristics
>>> HomeKit: [Client 5] Update Characteristics
RC > Identifying

`````

````
PROGRAM = main

EXTRA_COMPONENTS = \
    extras/onewire \
    extras/ds18b20 \
	extras/dht \
    extras/multipwm \
	extras/http-parser \
	extras/dhcpserver \
	extras/rboot-ota \
    $(abspath ../../external_libs/wolfssl) \
    $(abspath ../../external_libs/cJSON) \
    $(abspath ../../external_libs/homekit) \
    $(abspath ../../external_libs/wifi_config) \
    $(abspath ../../libs/led_codes) \
    $(abspath ../../libs/adv_button)

FLASH_SIZE = 8
FLASH_MODE = dout
FLASH_SPEED = 40

#HOMEKIT_SPI_FLASH_BASE_ADDR = 0xb0000 # 720896
HOMEKIT_SPI_FLASH_BASE_ADDR = 0x8c000

HOMEKIT_MAX_CLIENTS = 16
HOMEKIT_SMALL = 0
#HOMEKIT_OVERCLOCK = 1
#HOMEKIT_OVERCLOCK_PAIR_SETUP = 1
#HOMEKIT_OVERCLOCK_PAIR_VERIFY = 0

EXTRA_CFLAGS += -I../.. -DHOMEKIT_SHORT_APPLE_UUIDS -DWIFI_CONFIG_CONNECT_TIMEOUT=300000
EXTRA_CFLAGS += -DHOMEKIT_OVERCLOCK_PAIR_VERIFY
EXTRA_CFLAGS += -DHOMEKIT_OVERCLOCK_PAIR_SETUP

## DEBUG
#EXTRA_CFLAGS += -DHOMEKIT_DEBUG=1

include $(abspath ../../../esp-open-rtos/common.mk)

LIBS += m

monitor:
	$(FILTEROUTPUT) --port $(ESPPORT) --baud 115200 --elf $(PROGRAM_OUT)
`````

main.c only changed uint8_t device_type_static = 1 to uint8_t device_type_static = 5

homekit setup only added switch, no thermostat for adjust cool, heat...

+Other problem if i add extras/mbedtls to makefile
````
EXTRA_COMPONENTS = \
		extras/mbedtls \
`````
i got 
````
>>> HomeKit: [Client 4] Pair Setup Step 1/3
>>> HomeKit: [Client 4] Pair Setup Step 2/3
!!! HomeKit: [Client 4] Failed to compute SRP shared secret (code -2)
>>> HomeKit: [Client 4] Pair Setup Step 1/3
`````

kiralikbeyin, 2019-06-13:
no meaning adding or removing extras/mbedtls \
code is not stabil, generally i cant add but sometimes code works  

Never saw thermostat but only switch..

````
>>> homekit_server_on_pair_setup: [Client 5] Initializing crypto
>>> homekit_server_on_pair_setup: Free heap: 23404
>>> homekit_server_on_pair_setup: [Client 5] Using user-specified password: 111-88-111
>>> crypto_srp_init: Generating salt
>>> crypto_srp_init: Setting SRP username
>>> crypto_srp_init: Setting SRP params
>>> crypto_srp_init: Setting SRP password
>>> crypto_srp_init: Getting SRP verifier

>>> crypto_srp_init: Setting SRP verifier
>>> crypto_srp_get_public_key: Calculating public key

>>> send_tlv_response: [Client 5] Sending TLV response
>>> tlv_debug: Got following TLV values:
>>> tlv_debug: Type 3 value (384 bytes): \xA4h\xED\xC6\xE7\xFE\xAC)!\xAB*v\xE1\x939\x0F\xF0\xBB9\xD9&\xDD\xC1\x10e\x04L\xCAh\x91\x83\x84\xE5\xB1\xDE\x13\x90\x12~\xA4\x1Fk\x10\x98l\x93\xB3\xE0(\x0B\x0D\x1E\xFA\x1D\xE5\xB6\xD77\xF0\xBFj\x03\xA7w4\xEE\xC6f}\xC3\xF0\xE6\xB8H\xE6(&\xA0\xC2\xCD\xE08v^\xE4\xCF\x95\x081@\x04\xAC\xF5\xD0|\xE8~\x92(\x94\xAB\xF3\xC8\x10Rkw\xB7\x90-\xB4 p\xCD=\xFF$\xAA\xC4<M\x90\xEA>\xB5\x10T~\x85\xBF\x9Dx\xE0\xA1\xD5\xFE"\x11\x10\xAAH\xB7;\xD4\x89\xDE\x0Bd\xF4\x86Y\x0A\xEF\xF0=\x09\x06\xE3\xF8<\x06L\xE3U~}{\xFC39\x97\x1C;\xDEs.dc\xE5\x06;P\xC6\x1F\x08\xDE\xA6+\x95\x81\x17\xBC@\xD1 )\xBCX\xC5\xC4\x97\xAD\x09)\x8C_/\xC3\x18O\xB7\xD0\x1E\xE1\xE2\xFB\xA1m\xFB\xA6\x0D\x81\xA6\xA1\xDF\x9B\xF8\xBF\\\x11\x84\x0E\x90\x92V\xFB\xCD\xD5\xCD)\x14\x03\x06\xF9\x88E\x0BJ\x01h\xD2\xA1\xD2\xB1\x18\x9E\xD8\x06\x06x;j\xA6vm\x9B\xFA\xEF7\xCAp\x90\x94\xD2"9\x8Dg\xD53\x03\x06fC\x1D\x19\xCA\xD3E:\x13`\xF3;b\xE6\x9DNc*\xF0\xB8\x88\xE4\xE6\xAE\x0BxWN\xF2\xC0i\x13\xE9\xCE4v\xB4\x86\x04?[\x10\xDFu\x14\x1C2\x81\xD7LKI\xDE\x1De\x93\xFA5\x96\x11\xF0\xB1\xEE\x07Gus\x18?\xFE\x15\xD9W\x1D\xE5\xF4\x81\x9A\xBA\x0C\xA5e\x8A0\x04\xCC?\xB1,\xA9\x0E\x8C\x1D{\x89\x01\x0F)e\xF4
>>> tlv_debug: Type 2 value (16 bytes): \xC0m\xF17\xCEMf\xFE:f\x84\xD5\xB6\xA9\xB3\x97
>>> tlv_debug: Type 6 value (1 bytes): \x02
>>> client_send: [Client 5] Sending payload: HTTP/1.1 200 OK\x0D\x0AContent-Type: application/pairing+tlv8\x0D\x0AContent-Length: 409\x0D\x0AConnection: keep-alive\x0D\x0A\x0D\x0A\x03\xFF\xA4h\xED\xC6\xE7\xFE\xAC)!\xAB*v\xE1\x939\x0F\xF0\xBB9\xD9&\xDD\xC1\x10e\x04L\xCAh\x91\x83\x84\xE5\xB1\xDE\x13\x90\x12~\xA4\x1Fk\x10\x98l\x93\xB3\xE0(\x0B\x0D\x1E\xFA\x1D\xE5\xB6\xD77\xF0\xBFj\x03\xA7w4\xEE\xC6f}\xC3\xF0\xE6\xB8H\xE6(&\xA0\xC2\xCD\xE08v^\xE4\xCF\x95\x081@\x04\xAC\xF5\xD0|\xE8~\x92(\x94\xAB\xF3\xC8\x10Rkw\xB7\x90-\xB4 p\xCD=\xFF$\xAA\xC4<M\x90\xEA>\xB5\x10T~\x85\xBF\x9Dx\xE0\xA1\xD5\xFE"\x11\x10\xAAH\xB7;\xD4\x89\xDE\x0Bd\xF4\x86Y\x0A\xEF\xF0=\x09\x06\xE3\xF8<\x06L\xE3U~}{\xFC39\x97\x1C;\xDEs.dc\xE5\x06;P\xC6\x1F\x08\xDE\xA6+\x95\x81\x17\xBC@\xD1 )\xBCX\xC5\xC4\x97\xAD\x09)\x8C_/\xC3\x18O\xB7\xD0\x1E\xE1\xE2\xFB\xA1m\xFB\xA6\x0D\x81\xA6\xA1\xDF\x9B\xF8\xBF\\\x11\x84\x0E\x90\x92V\xFB\xCD\xD5\xCD)\x14\x03\x06\xF9\x88E\x0BJ\x01h\xD2\xA1\xD2\xB1\x18\x03\x81\x9E\xD8\x06\x06x;j\xA6vm\x9B\xFA\xEF7\xCAp\x90\x94\xD2"9\x8Dg\xD53\x03\x06fC\x1D\x19\xCA\xD3E:\x13`\xF3;b\xE6\x9DNc*\xF0\xB8\x88\xE4\xE6\xAE\x0BxWN\xF2\xC0i\x13\xE9\xCE4v\xB4\x86\x04?[\x10\xDFu\x14\x1C2\x81\xD7LKI\xDE\x1De\x93\xFA5\x96\x11\xF0\xB1\xEE\x07Gus\x18?\xFE\x15\xD9W\x1D\xE5\xF4\x81\x9A\xBA\x0C\xA5e\x8A0\x04\xCC?\xB1,\xA9\x0E\x8C\x1D{\x89\x01\x0F)e\xF4\x02\x10\xC0m\xF17\xCEMf\xFE:f\x84\xD5\xB6\xA9\xB3\x97\x06\x01\x02
>>> homekit_client_process: [Client 5] Finished processing

>>> homekit_client_process: [Client 5] Got 587 incomming data
>>> homekit_server_on_pair_setup: Pair Setup
>>> homekit_server_on_pair_setup: Free heap: 23192
>>> tlv_debug: Got following TLV values:
>>> tlv_debug: Type 6 value (1 bytes): \x03
>>> tlv_debug: Type 3 value (384 bytes): `\xA2\xD6R\xC5\x12v\xFD\x021DA\x19y\xF0\x16\xE5\x99\xB7\x80\x83\xA3$\x1E\xA8a\xD4\xCA\xB8\xDA+f[\xD0\xB5\x11m\xBE\xD1n\x0C\x15\xD2\x84\xB9a6\x15\x01+\xDFx\x96{-R\x05S\xF4\x9E\xAB\xD9\x02\x00\xF1\xC6b\x96j\xB3\x9B\xB2\xDF\x94\xF4roX\xCE\x03\xEC\xA2\x9E\xD2\xF3\xD7Q\xD4?\x94\xD6\xE1\x16\xBF\xA6\x06\x9DnxZ\xA2\x1C\xEAV\xC0]\xB1*D\xAAB\xC7\xD6\xB2\xEE\x9A\xD9\xEES\xAF\xA7GN\x8D\xED\xAD/\xD9\x84\x8D\x97\xFF.\xCA54H_\xED\x8A\x0Ap\x9A\xEF\xD3\x8D\x1F'_ \x13\xCB\xFB\xF3\xCF\xF5s]\x82)\xCE\x12\\\xB5\xE3=\xA2\x1B5O\x9B>\x97\x04\xAC\xDDE>\xD42L\xCC3D<\xEC\x94\xBC\xCE\x88r\xEDO\xC4\xEA\xEF\x90t\x862\xFD-4Ca&\xCCc\xC2\xD4?\x18\x19\xAC\x93\x0A\x10Q\xB9\x05\x0A\x8F)\\\x0A\x93q\x08*\xF3\xF3M\xE8\xA9[cz\xA7~u\x11\xBAn\x98:"64\xE90sU\xDF@\xA2\xD4\xD3b\xD1<a\xF8\x192\x13\xC8\xA9\x05\xCE\xE5\xC7{d\xAA\\V\xC8\xD5\x08\xF7E\x99\xA3\xF7i\x0A\xB2#\x1F\xA6P\xA2Q\xB8be\x19\x0E\xBA\x9B\xE9\x86\x83s\x888\x95\x1Fd\x90\xCA!\x92Rv\x04K\xFB\x82R\x1C\x1DZ\xE0\xA6\xBF^=\x19f\xC1\x888\x12\x93\x80\x0D\x92\xC8\xDD\x02\xFB\xFDV/o%\xF9\xF0jDQT*\xF1\x13\x05\xA5\x1F\x8F\x18\x93\xE7\xA9\x0C\xFF\xA0\x91\xE1\xCB3\xA4sm\xDD\xF0\xC5[J\x0B\xBF\x95\x17\xE0
>>> tlv_debug: Type 4 value (64 bytes): \x14\x05:\xDE(\xB8\xD7\xEDMt\x9D2\x10\x0B\x9A;\xA1\xC6d\x96\xD3"\x83\xBA\xC4\xE1\xB7|4F,>\xB0h\xCF\xBD\x8C\x10\xD8\x01?\x03-\x0A\x90\xEC2\x8B\xBAk\xAD\x0C\xE9\xA4\xB0\\U9q\x12\x01\xD0\x18\x01
>>> HomeKit: [Client 5] Pair Setup Step 2/3
>>> homekit_server_on_pair_setup: Free heap: 22624
>>> homekit_server_on_pair_setup: [Client 5] Computing SRP shared secret
>>> homekit_server_on_pair_setup: Free heap: 22624

>>> homekit_server_on_pair_setup: [Client 5] Verifying peer's proof
[... cut ...]
>>> homekit_server_on_update_characteristics: Free heap: 25480
>>> homekit_server_on_update_characteristics: [Client 4] Processing element {
"aid":	1,
"iid":	7,
"value":	true
}
>>> process_characteristics_update: [Client 4] Updating characteristic 1.7 with boolean true
RC > Identifying
>>> homekit_server_on_update_characteristics: [Client 4] There were no processing errors, sending No Content response
>>> client_send: [Client 4] Sending payload: HTTP/1.1 204 No Content\x0D\x0A\x0D\x0A
>>> homekit_client_process: [Client 4] Finished processing

````

RavenSystem, 2019-06-14:
Selecting Thermostat device type works for all people. And selecting any other accessory type too. There is not any issues about that. 

Other thing: RavenCore and esp-homekit library use WolfSSL instead mbedtls, maybe that is your problem. I don’t understand why you want to use mbedtls instead. 

And finally, changing pairing code is not allowed. Maybe a good practice is to read the license before: https://github.com/RavenSystem/esp-homekit-devices/blob/master/LICENSE

closed 2019-06-14 by RavenSystem