RavenSystem/esp-homekit-devices #411: HAA shelly1 stateless switch JSON config

gijsstegehuis, 2019-06-04:
Hi @RavenSystem,

I am trying to get a stateless switch running using HAA, but i can't get it to respond to the external button. This is the configuration i am using:
```
{
  "a": [
    {
      "t": 3,
      "b": [
        {
          "g": 5,
          "t": 1
        },
        {
          "g": 5,
          "t": 2
        },
        {
          "g": 5,
          "t": 3
        }
      ]
    }
  ]
}
```

A [per the docs](https://github.com/RavenSystem/esp-homekit-devices/wiki/HAA-Templates#shelly1-switch-with-external-toggle-note-that-t1-is-not-present-because-is-the-default-option) it looks like gpio should be set to 5, but it doesn't work for me.
Any thoughts on this?

RavenSystem, 2019-06-04:
With Shelly you must disable button pull-up resistance.

gijsstegehuis, 2019-06-05:
That works perfectly. Thx!

closed 2019-06-05 by gijsstegehuis

lucaamo, 2019-07-09:
Can you share the correct json to use. Thanks

gijsstegehuis, 2019-07-10:
@lucaamo see below:
```{
  "a": [
    {
      "t": 3,
      "b": [
        {
          "g": 5,
          "t": 1,
          "p": 0
        },
        {
          "g": 5,
          "t": 2,
          "p": 0
        },
        {
          "g": 5,
          "t": 3,
          "p": 0
        }
      ]
    }
  ]
}```

lucaamo, 2019-07-10:
[image]

it is normal that works like that?

mafyata, 2019-07-10:
What app do you use? I am talking for the app of the screenshot.

lucaamo, 2019-07-11:
EVE

gijsstegehuis, 2019-07-11:
@lucaamo can you send an Apple Home app screenshot of the same button?
I have no clue what it's supposed to look like in the Eve app.

lucaamo, 2019-07-13:
[image]

lucaamo, 2019-07-13:
I am getting:
- a single press when I do a double press, 
- a double press when I do a  triple press
- a long press when I do a single press

gijsstegehuis, 2019-07-13:
@lucaamo yes. that's correct.

HAAnewbie, 2020-02-02:
I've tried configuring the Shelly1 with the JSON string that was proposed above, but I'm unable to press the button in the iOS Home app. What would be the correct string so that I can either use the momentary physical push button connected to the SW pin on the Shelly (GPIO5) or use the Home app to trigger that button? Action is then to momentarily short pins I and O on the Shelly1 (GPIO4). I'm basically trying to control a garage door by connecting the current push button to Homekit. Right now, 1 press on that button activates the door, another press stops it, another press activates it again but in the other direction, etc and all this happens in the garage actuator. I just need the Shelly to short O and I whenever the physical button is pushed or whenever I push the button in the Home app.

Quoshak, 2020-05-28:
Hi,

I am also trying to make shelly1 to work as a push button. Json that was published by @gijsstegehuis in this topic is not working for me. 
I can see button in Home App, and I can setup actions, but nothing is happening when I push my switch or press button in Home App.

Can somebody help? I am not sure what I am missing.

gijsstegehuis, 2020-05-28:
@Quoshak the config might be outdated by now.
Any Idea @RavenSystem?