RavenSystem/esp-homekit-devices #414: No AP mode available after v0.90 fw flash; Sonoff

vojtechtambor, 2019-06-07:
After FW flashing, I get no AP available so that i can proceed with setup

Sonoff Basic unit
Flash erased OK
FW uploaded OK

kept the sonoff connected to USB via serial adapter for power

the unit should create an access point but none is available

when i tried the Tasmota FW all worked OK, so the hardware should be fine

thanks!

dennild, 2019-06-07:
i have had the same issue, i fixed it by doing a power cycle,
Also did you use the Ravencore main.bin file or the OTA one?

vojtechtambor, 2019-06-07:
i used the main.bin, not the OTA.

tried switching it off an on couple times, no change

what is the approx time required to see the AP after bootup?

mafyata, 2019-06-07:
I use ota and I can see the AP after 5-10 seconds. Sonoff must be disconnected from usb to serial adaptor and must be connected to 110 or 220 AC.

vojtechtambor, 2019-06-07:
Thanks! that helped, once i connected the sonoff unit to 220AC, it created the AP immediately.

closed 2019-06-07 by vojtechtambor