RavenSystem/esp-homekit-devices #419: RavenCore support SONOFF Mini?

pilot1981, 2019-06-10:
Hi,
Ravencore support SONOFF Mini Two Way Smart Switch For Smart Home Can Work With An External Switch?

please, see this link:

https://www.banggood.com/SONOFF-Mini-Two-Way-Smart-Switch-For-Smart-Home-Can-Work-With-An-External-Switch-p-1484758.html?utmid=9067&cur_warehouse=CN

P.S: SONOFF Basic R3 is arriving...

RavenSystem, 2019-06-10:
It works in the same way as a Sonoff Basic. I can add it knowing its GPIO map.

CooperNick, 2019-07-04:
Hi
I received Sonoff Minis
What info do you need about GPIO?
Regards

RavenSystem, 2019-07-04:
Try RavenCore first.

closed 2019-07-20 by RavenSystem

den23rus, 2019-08-03:
> Hi
> I received Sonoff Minis
> What info do you need about GPIO?
> Regards

Hi please tell me do u try flash sonoff mini by ravencore? what about result? work?

CooperNick, 2019-08-03:
>> Hi please tell me do u try flash sonoff mini by ravencore? what about result? work?

Not yet, will try to make pogo-pin connector.

brunokelo, 2019-08-14:
@RavenSystem 

Hello, is it available?

basically the same as the basic sonoff except that GPIO4 is a switch input.

skv6, 2019-10-02:
Sonoff mini work but like sonoff basic - only remote control - external toggle not supported like in original software.

JefersonSchramm, 2020-01-26:
Sonoff Mini with external toggle and built-in button configure to enter setup mode:
{"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"0":{"r":[{"g":12}]},"1":{"r":[{"g":12,"v":1}]},"b":[{"g":0},{"g":4},{"g":4,"t":0}]}]}