RavenSystem/esp-homekit-devices #420: Relay does not trigger

RajuRah, 2019-06-11:
Thanks for the amazing work!!!! On sonoff basic works perfect, but thats not enough for me, i can not toggle sonoff and manual switch .... so i tried on shelly 1:

but i have problem with Shelly 1:
   - i was able to flash and get into homekit - works fine with setting up / finding device in homekit etc. also the status on Homekit app shows correctly
   - but the Light turns not ON or OFF - no sounds from relay
   - the wiring should be fine, because I used the same setup to connect to the shelly APP before, which worked fine ... and i was able to toggle

Any idea what can be the problem? i tried with ravencode and HAA flash ...
then tried with tasmotag to check and did not work at all ... any help will be great. Thanks.

RavenSystem, 2019-06-12:
You must configure your device. Read Wiki, please.

closed 2019-06-12 by RavenSystem

RajuRah, 2019-06-13:
> You must configure your device. Read Wiki, please.

Thank you for the information, I am searching in WIKI for an 1/2 hour ... and still not sure about where i can configure the device ... do i have to make the change in EVE app?

mafyata, 2019-06-13:
Yes , you must configure it in eve app.

RajuRah, 2019-06-13:
> Yes , you must configure it in eve app.

Thank you very much, it works perfect ... :)