RavenSystem/esp-homekit-devices #422: Question: is Sonoff BasicR3 Supported?

dkossman, 2019-06-13:
Does this firmware/installation process work with the new Sonoff BasicR3?  https://sonoff.itead.cc/en/products/sonoff/sonoff-basicr3

My understanding is that this version of the Basic Sonoff switch includes a REST API which could be used e.g. with homebridge , but it still requires use of the Sonoff eWeLink app and still "calls home" to the Sonoff back end.

thanks

RavenSystem, 2019-06-13:
With HAA, sure. With RC, I did not test it, maybe somebody will do it.

RAZRu, 2019-06-29:
> Does this firmware/installation process work with the new Sonoff BasicR3? https://sonoff.itead.cc/en/products/sonoff/sonoff-basicr3
> 
> My understanding is that this version of the Basic Sonoff switch includes a REST API which could be used e.g. with homebridge , but it still requires use of the Sonoff eWeLink app for initial setup and still "calls home" to the Sonoff back end.
> 
> thanks

Works fine with default settings RC(dev type1 board type 1)

closed 2019-06-29 by RavenSystem