RavenSystem/esp-homekit-devices #423: Problem has stop garage door

pilot1981, 2019-06-13:
Hi,
All worked fine with my two garage door and ravencore 0.8.7.

Today I tried to update to 0.9.2 version with has stop step on, but I notice when I send open command, sonoff start relay and start again relay after about 1 second, so garage door stop and don’t open....

Config of has stop step is always on...and before upgrade all work fine....

My configuration error or a bug?

RavenSystem, 2019-06-13:
Bug, sure. Disable "has stop step" until I release a fix. Sorry.

RavenSystem, 2019-06-13:
Fixed in 0.9.3.

closed 2019-06-13 by RavenSystem

pilot1981, 2019-06-14:
I updated to 0.9.3: with has stop step off, garage door stop if I press again button: it’s ok for you?

RavenSystem, 2019-06-14:
@pilot1981 Ok, now enable "has stop step" and see if your garage door change way of motion when it is moving and you press HomeKit button. It must stop and then move again.

skanico, 2019-06-17:
Hello, I updated to 0.9.6 and the "has stop" function still doesn't work as expected (not really an issue for me though, as I don't use HomeKit button to invert the motion) - physical buttons still do the work -.
Everything else works fine for me.

pilot1981, 2019-06-23:
Hi,
Sorry for delay in my answer.

I tried with and whiteout has stop step feature but behavior don’t change: with both configuration, when garage is opening or closing and I use again button from app, garage door stop and when I click again garage door start again to opening/closing...

What is it wrong?

pilot1981, 2019-07-04:
@ravensystsem Sorry, but also in 1.0.0 RC version has stop step feature don’t change behavior of device: it seem to do nothing...

pilot1981, 2019-07-08:
@RavenSystem Sorry, but also in 1.0.0 RC version has stop step feature don’t change behavior of device: it seem to do nothing...

RavenSystem, 2019-07-08:
In my case it works perfectly, when door is opening and when door is closing, with and without open sensor.

pilot1981, 2019-07-08:
yes, also for me works fine, but if I setup "has stop step" to NO, it works in same way...very strange...

RavenSystem, 2019-07-08:
Can you explain with detail your issue??

pilot1981, 2019-07-08:
with setup "has stop step" to NO, I open garage door successful, during garage door opening I click again on same button to stop garage door, and garage door stop  successful...

If I understood good: with setup "has stop step" to NO, garage door couldn't stop during opening/closing operation...

pilot1981, 2019-08-11:
Hi @RavenSystem,
I would confirm bug.

With has stop step enabled, if garage door is opening and I send command garage door stop correctly but after few second my sonoff send again signal to garage door so garage door start again to move...

I don’t think this is correct behavior...I would use this feature to stop garage door in any position I prefer...

pilot1981, 2019-08-11:
I have only close sensor.

Question: this feature require also open sensor?

pilot1981, 2019-08-11:
Please @RavenSystem can you open again it and correct this bug?

RavenSystem, 2019-08-25:
"has stop step" works perfectly. This option is not for stop your door. That option is not implemented in Apple HomeKit; it is to avoid to stop it pushing action button twice instead one.