RavenSystem/esp-homekit-devices #424: th no temp or humidty - thermostat accessory

kiralikbeyin, 2019-06-14:
th added like a single switch, there is no thermostat accessory
0.9.3

i only changed main.c -> uint8_t device_type_static = 5,

RavenSystem, 2019-06-14:
You must use EVE or any other third party HomeKit App to configure “Device type”.

kiralikbeyin, 2019-06-14:
@RavenSystem thanks, i used your lib 10-12 months ago and it is very different now.
So i will upload only a single firmware to all my sonoffs and set them as i want, all in one thats cool.

I checked other devices like kellyTron HAA but didnt understand are they prototypes of RavenCore?

How can i directly set accessories like before (without eve https://github.com/RavenSystem/esp-homekit-devices/releases/tag/0.1.2) I want to keep codes very simple..

closed 2019-06-14 by kiralikbeyin

RavenSystem, 2019-06-14:
> @RavenSystem thanks, i used your lib 10-12 months ago and it is very different now.
> So i will upload only a single firmware to all my sonoffs and set them as i want, all in one thats cool.

Yes, now RavenCore is a very refined firmware with many options keeping low resources usage.

> I checked other devices like kellyTron but didnt understand are they prototypes of RavenCore?

KellyTron was a first approach to Shelly devices, but now it is discontinued and RavenCore is able to manage Shelly devices. I will remove KellyTron code soon because is very outdated.

> How can i directly set accessories like before (without eve https://github.com/RavenSystem/esp-homekit-devices/releases/tag/0.1.2) I want to keep codes very simple..
> edit i think HAA is for directly setting accessories

With RavenCore you must use EVE or any other their party Homekit App to configure it. With HAA you can build a custom Homekit firmware using a JSON string, but thermostat actually is not supported... yet.

mafyata, 2019-06-14:
Can you tell me , another homekit app like eve that I can config ravencore?

St0nehead, 2019-07-06:
https://apps.apple.com/kz/app/controller-for-homekit/id1198176727

Using this one about a year now...Very satisfied..