RavenSystem/esp-homekit-devices #425: External toggle for t1 3ch

xtremedays, 2019-06-15:
Could you add an external toggle option to the 3ch switch please (device type 11). (as with the 1ch and 2ch switch options.

RavenSystem, 2019-06-20:
With RC no, but you can do the work with HAA. What GPIOs are you using for the external toggles?

xtremedays, 2019-06-20:
0, 10 and 9. I got it working as a push button switch on those GPIO's, just can't work out how to do the toggle switch

RavenSystem, 2019-06-20:
Put here your JSON, and I will help you to convert buttons to toggles.

xtremedays, 2019-06-20:
{"a":[{"r":[{"g":12}],"b":[{"g":0}]},{"r":[{"g":5}],"b":[{"g":10}]},{"r":[{"g":4}],"b":[{"g":9}]}]}

RavenSystem, 2019-06-20:
{"a":[{"r":[{"g":12}],"b":[{"g":0},{"g":0,"t":0}]},{"r":[{"g":5}],"b":[{"g":10},{"g":10,"t":0}]},{"r":[{"g":4}],"b":[{"g":9},{"g":9,"t":0}]}]}

To create a toggle, you must declare 2 button types, 0 and 1. Type 1 is the default, and it is triggered when you up button; and type 0 is triggered when you down button.

xtremedays, 2019-06-20:
ok. Will it cause issues with gpio0 used as a toggle and resetting, thinking thats its being held down?

RavenSystem, 2019-06-20:
No, HAA can not be reseted holding a button. Well, HAA can not be reseted in any way, only can enable setup mode, but nothing of your setup is lost. And the only way to do is this: https://github.com/RavenSystem/esp-homekit-devices/wiki/Home-Accessory-Architect#entering-setup-mode

Problem using GPIO 0 as a toggle is that if you power your device and GPIO 0 is in low state, your device will enter in flash mode.

xtremedays, 2019-06-20:
Ok, I will have to see if another GPIO can control switch 1. I got the GPIO's from here [https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-T1](url).

Thanks for your assistance!

RavenSystem, 2019-06-20:
I think that you can use GPIO3, that is RX pin, instead GPIO 0.

xtremedays, 2019-06-21:
All sorted, thanks for that. Can use GPIO3 as button 1 as well.

closed 2019-06-21 by xtremedays