RavenSystem/esp-homekit-devices #426: HAA v0.0.5 frozen when LED enabled

i3laze, 2019-06-16:
Doing my first steps with HAA - very promising.

I’ve just enabled Basic R2 with a relay and a button - was working great.

Then I’ve added a LED (seems, I forgot to invert it) and device is irresponsible now:
LED glows, button doesn’t trigger, not responding in HomeKit.

Please, give a try: 
{"c":{"l":13},"a":[{"r":[{"g":4}],"b":[{"g":5,"p":0},{"g":5,"t":0}]}]}

RavenSystem, 2019-06-16:
Can you post logs, please?

i3laze, 2019-06-16:
Hi, yeap, but during the day.

RavenSystem, 2019-06-16:
> P.S. iOS Safari inputs italic double quotes “ (not ") into JSON field on setup webpage, can you fix this?

Please, open a new issue about this because I can forget. :)

RavenSystem, 2019-06-16:
For Sonoff is better to enable internal pull-up resistors (In Shelly it must be disabled). But if your device doesn't respond, you must erase and flash it again. Omitting LED inverted should not have negative effects, only it will work in the opposite way.

i3laze, 2019-06-16:
UPD: Cold start made it working.. with a configuration older, then I remember I've set.
Sorry, no traces left in log.

Hovewer, I guess.. the cause might be this one:
`HAA` ! ERROR: Invalid JSON
though the device may have stuck with LED enabled in setup mode.

I suggest when such JSON startup error occurs some LED signaling should be in place, like S-O-S in morse code.

P.S. Here's most basic template for Sonoff with button and switch:
{"c":{"l":13,"i":1},"a":[{"r":[{"g":12}],"b":[{"g":0,"t":0}]}]}
I figured out GPIO from a device library at Tasmota Wiki here:
https://blakadder.github.io/templates/sonoff_basic.html

i3laze, 2019-06-16:
I've also noticed setup page doesn't do JSON validation
(however device does it on boot, which may be not enough)..

And HAA doesn't format JSON input field to a compact one-liner after Save. It might be visually nice.

i3laze, 2019-06-17:
UPD: Another Cold start: now device is in "Invalid JSON" + SoftAP mode.
If I hadn't have Serial attached I wouldn't have known I should go look for a SSID. 

Definitely some GPIO13 signaling scheme required.

RavenSystem, 2019-07-20:
It is not possible, because LEDs are defined in JSON.

closed 2019-11-07 by RavenSystem