RavenSystem/esp-homekit-devices #429: HAA with a reed sensor

i3laze, 2019-06-17:
I'm adding a reed sensor (a water level float sensor actually) to Sonoff Basic R2 GPIO3 to trigger a Dummy Switch on HAA.

As simple as it is, I can't make the button work in push-and-hold mode:
GPIO3 connected to GND = Dummy Switch On
GPIO3 disconnected = Dummy Off

Should I disable the pullup?
_{"a":[{"b":[{"g":13,"t":1}]}]}_

i3laze, 2019-06-18:
UPD: Trial and error..) Inverted mode it is.
Here it's working:
`{"c":{"l":13,"i":1},"a":[{"r":[{"g":12}],"b":[{"g":0,"t":0}]},{"b":[{"g":3,"i":1},{"g":3,"t":0}]}]}`

i3laze, 2019-06-18:
Not exactly what I wanted.
Indeed opening/closing GPIO3 toggles state. But..

State doesn't follow GPIO high/low: It can be toggled in HomeKit to opposite and it can't be queried on startup.

i3laze, 2019-06-19:
I've solved it with RavenCore: Lock Mechanism (Type 13) as it has the separate sensor.
But, mate.. GPIO4? :(  have to solder it.
I wish GPIO was configurable in settings..

RavenSystem, 2019-06-20:
Only a thing: this JSON is not right:
{"c":{"l":13,"i":1},"a":[{"r":[{"g":12}],"b":[{"g":0,"t":0}]},{"b":[{"g":3,"i":1},{"g":3,"t":0}]}]}

It should be:
{"c":{"l":13,"i":1},"a":[{"r":[{"g":12}],"b":[{"g":0,"t":0},{"g":3,"i":1},{"g":3,"t":0}]}]}

And I see that GPIO 0 is a push button, so I suggest to use type 1 instead type 0, that will trigger button when you release it. That is the common normal way to work of buttons.

i3laze, 2019-06-21:
There’s a bunch of videos and wiki [Wiki article](https://github.com/arendst/Sonoff-Tasmota/wiki/Buttons-and-Switches) about Tasmota SwitchMode (7 options) and SwitchTopic (2 options) combinations..

Here in HAA you’ve got about same combinations less documented AND a raw JSON to stumble in it :(

1. Accessory Sensor still required, as Switch can be messed/toggled by user within HomeKit.
2. Regarding online configuration tool.. may be it’s better to hide JSON under the table get back to Eve.app config as RC does?

Accessory count? 1/2/3/4
1st Accessory type: 1-15
1st Relay GPIO: 1-15
1st button GPIO: 1-15
Etc..

Absolutely same configurator ought to be created online..so why not put it inside Eve instead..

RavenSystem, 2019-06-28:
> HAA Accessory: Sensor still required, as Switch ins't queried on startup and can be messed/toggled by user in Home.app.

Sensors are not implemented yet. 

> May be accept in JSON long words like ‘gpio’, ‘invert’, ‘relay’, ‘button’? They are a bit more vivid.

No, because flash storage is very limited. In a future, you will can create JSON string using an external app/web more comprehensive. 

> Regarding online configuration tool.. may be it’s better to hide JSON under the hood and get back to Eve.app config as RC does?

No, because each HK characteristic (that is EVE uses) needs 96bytes of RAM with its declaration, and then more RAM to use it. Moving to a JSON model is to preserve RAM and flash, so you can create more complex accessories.

closed 2019-06-28 by RavenSystem