RavenSystem/esp-homekit-devices #430: iOS 13

netRunner0, 2019-06-18:
Hi, no settings(
[image]
[image]

archiolochos, 2019-06-18:
same here. I am using ios13beta2.i am sure that there were settings in ios13beta1.

claire078, 2019-06-19:
You must set it to off and on again.

i3laze, 2019-06-19:
Cool house у тебя!
I guess that's an Eve.app bug..  to be fixed when iOS 13 released in autumn.

caligari74, 2019-06-20:
This didn't work for me on beta2.
I slid "Show Setup" to off and back to on.  The settings still don't show.

RavenSystem, 2019-06-20:
iOS 13 is beta, so testing actual EVE app with it not make sense.

closed 2019-06-20 by RavenSystem