RavenSystem/esp-homekit-devices #431: DHT11/22 as Sensor in Home App

Codkit, 2019-06-19:
Hi José

Thanks for all your time and efforts ,this really improve my home automation system.
Right now i have a question if its possible to use dht11/22 as a sensor in home app so it can trigger remotely other homekit devices in the same network ?
Currently if i create a new automation in home app,the “sensor” is greyed out (device type 5/6/9/ in Eve).
Thanks again for all your contribution !

Regards
Cod

mafyata, 2019-06-19:
IFTTT

claire078, 2019-06-19:
Eve app

Codkit, 2019-06-20:
Thanks mafyata and Claire, can you show me how in ifttt or eve ? Thank you

St0nehead, 2019-07-06:
https://apps.apple.com/kz/app/home-3/id995994352

You can add new automation with your “greyed” sensors within this app. Then, your automations appears in Home.app

Codkit, 2019-07-07:
I wish the app is free... anyway thanks for the link

closed 2019-07-20 by RavenSystem