RavenSystem/esp-homekit-devices #432: HAA toggle switch

xtremedays, 2019-06-19:
Just trying to set up HAA to be used with a toggle switch. I've tried using "t":5, but doesn't seem to work..... I can't trigger it off the switch, but can using the app.

json for a basic switch with toggle switch....
{"a":[{"r":[{"g":12}],"b":[{"g":0,"t":5}]}]}

is this right?

i3laze, 2019-06-19:
Hi!

g:0 stands for GPIO0..  What device you're using?
Toggle switch is kind of one on the walls? Up/Down?

xtremedays, 2019-06-19:
yea sonoff basic. (and t1 3 gang, wanting to convert it to normal switches). 

maybe i should try GPIO2... 

Yes toggle switch is your normal light switch, on/off, not momentary.

RavenSystem, 2019-06-20:
@xtremedays What do you want to do exactly with GPIO 0?? Ina Sonoff Basic, GPIO 0 is attached to built-in button, and configure it as a toggle is not a good idea. maybe this can help you: https://github.com/RavenSystem/esp-homekit-devices/wiki/HAA-Templates#sonoff-basic-switch-with-uart-output-status-led-working-button-and-external-toggle-connected-to-gpio-14

xtremedays, 2019-06-20:
@RavenSystem So that was a mistake, as i3laze pointed out... i want to use a toggle switch with either a sonoff basic (which i have done through RavenCore) or sonoff t1 US. 

I somehow missed that particular template has a toggle switch! I will try that next time I get a chance. Thanks

closed 2019-06-21 by xtremedays