RavenSystem/esp-homekit-devices #433: Question HAA - Which Switch & postion

kiralikbeyin, 2019-06-19:
@RavenSystem Where can i see when a switch postion changes? I need to put them to variables.

i3laze, 2019-06-19:
Mate, variables.. that sounds like you need Tasmota stuff.

Describe your end goal.

RavenSystem, 2019-06-20:
@kiralikbeyin A switch without output GPIOs is a dummy switch, and it is only visible in HomeKit. I agree with @i3laze, maybe a description of your end goal could help.

kiralikbeyin, 2019-06-20:
@i3laze @RavenSystem 
When i change position of dummy switch, i want to send info to a server. (I added a fake gpio to dummy switch so i can understand which dummy switch pressed like this:
I add pushToServer function to "void hkc_on_setter(.."
```
...
        const bool relay_value = value.bool_value ^ inverted;
        gpio_write(gpio, relay_value);
        printf("HAA > pushToServer Output GPIO %i -> %i\n", gpio, relay_value);
        pushToServer(gpio,relay_value);
}
```

Now i am trying : Push notification from my server to dummy switch.

dummy switch with a fake gpio (50) and a fake button gpio (100)

if i write something like digitalWrite(100,true) , will it be ON?

kiralikbeyin, 2019-06-21:
Setup:

````
{
  "a": [
    {
      "t": 1,
      "r": [
        {
          "g": 50
        }
      ],
      "b": [
        {
          "g": 100,
          "t": 1
        }
      ]
    },
    {
      "t": 2,
      "r": [
        {
          "g": 51
        }
      ],
      "b": [
        {
          "g": 101,
          "t": 1
        }
      ]
    }
  ]
}
````

Trigger:

````
           gpio_write(100, true);

           vTaskDelay(100 / portTICK_PERIOD_MS);

           gpio_write(100, false);
````

Simply i want to notify the switch to on/off by gpio_write of button but didnt work.

How can i manually change position of an accessory with a trigger code?

kiralikbeyin, 2019-06-21:
add to adv_button.h
````
void dummy_button(const uint8_t used_gpio) ; //fake button to trigger the accessory ---> format ----> {"a":[{"t":1,"r": [{"g":100}], "b": [{"g":50,"t": 1}]}]}
````

add to adv_button.c
````
void dummy_button(const uint8_t used_gpio) {  
//fake button to trigger the accessory {"a":[{"t":1,"r": [{"g":100}], "b": [{"g":50,"t": 1}]}]}
push_up(used_gpio);
}
````

usage
````
           char gpiofinder [4] ;
...
...
           int ok =  atoi(gpiofinder)  ;
           printf("Trigger to fakegpio %i\n",ok );
           dummy_button(ok);
````

created pull requests

closed 2019-06-21 by kiralikbeyin