RavenSystem/esp-homekit-devices #434: Garage Doors  2 relays

okc-esp, 2019-06-19:
Hello. Available: engine for the gate (Nice RB600) and NodeMCU v3 board stitched RAVENCORE (v0.9.5).

When connecting NodeMCU to the engine from the gate, a problem was detected, the engine has 3 connectors: Open, Close and Stop. The firmware is not designed for this connection. The firmware supports only one relay, which is responsible for opening and closing the gate.

If I connect the Relay to the “open” connector, then there is no possibility to close the gate. When you click on the "close" icon in HomeKit, it works out the relay, thereby closing the "Open" connector on the gate.

Please consider connecting 2 relays. One relay will be responsible for opening, and the second for closing the gate.
Or the ability to disable relay triggering when you click on the “Close” icon in Homekit. For the reason that the gate has time, after which they close themselves.

i3laze, 2019-06-19:
For the time being you might go with Window Covering (type 12) and Sonoff Dual (NodeMCU might fit too).
This mode also has inching and Relay1 and Relay2 are in interlock mode.

okc-esp, 2019-06-19:
i3laze, thank you for the answer.
I am currently using Window Covering. But now there are not enough sensors.

Tasmota firmware uses MQTT, but I want to work with the Native device.
Just this gives RavenCore, in my opinion one of the best firmware.
Regarding the change in GPIO, I do not need this.
I just now lack the second relay in this firmware. Therefore, I asked to consider the possibility of adding this functionality.

i3laze, 2019-06-20:
Have a great news for you👍 
I’ve taken a closer look at your RB600 manual: there’s P.P. input, which allows step-by-step control using one input. It’s also programmable from motherboards’ side (check manual p7.4, layer L2).

Good luck with tuning the door up to RavenCore!

P.S. You’ll still require an external reed sensor to detect closed state, like [this one](
http://s.aliexpress.com/zmuuUBb6).

RavenSystem, 2019-06-20:
@okc-esp You are looking for a complete garage door driver, like RavenCore window covering setup, but this is not the case. RC Garage door accessory type is focused to add Homekit to your actual garage door driver, not replacing it.

okc-esp, 2019-06-20:
Thank you so much, i3laze for the help.
I studied the documentation for the motor from the gate, and missed the P.P. connector.
When I saw the OPEN and CLOSE connectors, I stuck to them.

RavenSystem, I really liked your work.
Having studied my engine from the gate badly, I had a problem. Since there was one relay. And it seemed that 2 relays are required for normal operation.
Maybe someone in the future will need 2 relays. Since there will be no one universal connector on the engines from the garage door. Therefore, please consider adding a second relay to the garage door.

In my case, the problem is solved. And I do not need this functionality.
Thank you for the great work that helps other people make their home a bit automated.

closed 2019-06-20 by okc-esp