RavenSystem/esp-homekit-devices #435: HAA Crash, inverted mode

i3laze, 2019-06-19:
I've tried relay inverted mode on Sonoff SV and got a boot loop:
`{"c":{"l":13,"i":1},"a":[{"r":[{"g":12},{"i":1}],"b":[{"g":0,"t":0}]}]}`
```
AA > Enable LED GPIO 13, inverted=1
HAA > Creating Acc 0, type=1
HAA > Enable button GPIO 0, type=0, inverted=0
HAA > Enable output GPIO 12
HAA > Output GPIO 12 -> 0
Fatal exception (28): 
epc1=0x4021714b
epc2=0x00000000
epc3=0x00000000
excvaddr=0x0000001c
depc=0x00000000
excsave1=0x4021714b
Registers:
a0 4021714b a1 3fff2b70 a2  00000000 a3  00000000
a4  00000000 a5  00000001 a6  00000001 a7  0000001a
a8  0000001a a9  000000b0 a10 60000000 a11 0000000a
a12 00000001 a13 3fff44d8 SAR 00000014
```

RavenSystem, 2019-06-20:
Your JSON is not right:
{"c":{"l":13,"i":1},"a":[{"r":[{"g":12},{"i":1}],"b":[{"g":0,"t":0}]}]}

Should be:
{"c":{"l":13,"i":1},"a":[{"r":[{"g":12,"i":1}],"b":[{"g":0,"t":0}]}]}

i3laze, 2019-06-21:
Yeap, but my “valid JSON” made device unusable (loop every 3 sec), so you’d better catch it somehow someday.

closed 2019-07-20 by RavenSystem