RavenSystem/esp-homekit-devices #436: Garage door: Has stop step options..

i3laze, 2019-06-20:
Did thorough tests with Sensor Close NC off and on, Inching time 0,3;0,6; Working time 5;17.
Sonoff SV working just fine on v0.9.6, I'd say 95% fine.
 
However.. noticed a **bug**, which might (or might not) cause cascade/wrong behavior:
_(@RavenSystem, please confirm)_

With Has Stop Step On and NC both Off or On:
State: _Open_ -> Closing -> I click -> Opening back.. -> Open.
Device doesn't respect _"Has Stop Step On"_ and does 1 click only.

It still does two clicks when Closed -> Opening -> I Click -> Closing.

i3laze, 2019-06-20:
https://github.com/RavenSystem/esp-homekit-devices/issues/428#issuecomment-504132744

i3laze, 2019-06-20:
UPD: I suggest just fix it back.
_Nice_ doors have an option to configure it to RC default behavior.

RavenSystem, 2019-06-20:
@i3laze Thanks!

closed 2019-06-20 by RavenSystem