RavenSystem/esp-homekit-devices #437: Question: Water valve for Sonoff Dual

skipper79, 2019-06-21:
Hi!

I really like RavenCore solution. At the moment I uses the device types ‘Garage Door’ and ‘Water Valve’. The last one I used in combination with a magnetic valve, but the power consumption is quit high. So I want install a 2-way valve for opening and closing a valve. For that the the device ‘Window Covering’ is a nice solution with a Sonoff dual. But, for this I missies the type ‘Valve’. Is it possible to combinate ‘Water Valve’ and ‘Window Covering’ to make a toggle switch for opening and closing the water valve? Looking forward to your reply!

i3laze, 2019-06-22:
Actually, duplicate [https://github.com/RavenSystem/esp-homekit-devices/issues/369#issuecomment-495996616]

Got the same ball valve. And Window covering type currently works great.

Since v0.9.1 it even utilizes a button to
open (1 click) and close (2 clicks).

Timer to close can be set by automation (dummy might be required).

I think reprogramming it go Valve time is not that tricky - just separate “open” GPIO from “close” and ensure interlock.

RavenSystem, 2019-06-28:
Those water valves that require 2 relays will be supported, but not soon.

closed 2019-06-28 by RavenSystem

nov-, 2021-11-04:
Thank you for your work! It's super! Is it already supported? I mean sonoff with two relays and water valve.

i3laze, 2021-11-04:
Yeah, Sonoff dual with ball valve did the job great the whole summer.
I hooked a float sensor and had a pool automatically re-filled to the level every few mornings.

nov-, 2021-11-06:
> Yeah, Sonoff dual with ball valve did the job great the whole summer.
> 
> I hooked a float sensor to Duals gpio13 pin and had a pool automatically re-filled to the level every few mornings.

could you please share your json config for this?

i3laze, 2021-11-06:
Please review it at [online HAA configurator](url) and remove unnecessary features.

`{"c":{"b":[{"g":0,"t":5}],"o":0,"i":1},"a":[{"0":{"r":[{"g":12,"v":1,"i":27},{"g":13,"v":1},{"g":5,"v":0}]},"1":{"r":[{"g":5,"v":1,"i":27},{"g":13,"v":0},{"g":12,"v":0}]},"t":20,"b":[{"g":10,"t":1}],"s":0,"w":0,"d":0},{"t":7,"g0":[{"g":9,"t":0,"i":0}],"g1":[{"g":9,"t":1,"i":0}]},{"0":{"m":[[1,0]]},"f1":[{"g":9,"t":0,"i":0}],"t":1,"i":180,"b":[],"s":0}]}`

nov-, 2021-11-06:
> Please review it at [online HAA configurator](url) and remove unnecessary features.
> 
> Check your device pin-out. Here’s SonoffDual:
> 
> https://templates.blakadder.com/sonoff_dual_R2.html
> 
> 
> 
> Sonoff Dual with rotary ball valve connected to this pins:
> 
> 
> 
> - GPIO12 in inching mode to close valve for 27 sec (turns off GPIO5)
> 
> - GPIO5 to open for 27 sec (turns off GPIO12)
> 
> - GPIO13 - LED
> 
> - GPIO10 - Button
> 
> - GPIO9 - Floating leak sensor (type 20)
> 
> 
> 
> Extra virtual inching switch, which turns off valve 180 sec after floating sensor has raised above level. This adds some extra water so the valve won’t start too often.
> 
> 
> 
> `{"c":{"b":[{"g":0,"t":5}],"o":0,"i":1},"a":[{"0":{"r":[{"g":12,"v":1,"i":27},{"g":13,"v":1},{"g":5,"v":0}]},"1":{"r":[{"g":5,"v":1,"i":27},{"g":13,"v":0},{"g":12,"v":0}]},"t":20,"b":[{"g":10,"t":1}],"s":0,"w":0,"d":0},{"t":7,"g0":[{"g":9,"t":0,"i":0}],"g1":[{"g":9,"t":1,"i":0}]},{"0":{"m":[[1,0]]},"f1":[{"g":9,"t":0,"i":0}],"t":1,"i":180,"b":[],"s":0}]}`

Thank you very much!

nov-, 2021-11-07:
One more question 🙂 will it work without floating leak sensor (type 20)? Because I need just open/close my water valve and beautiful icon of valve instead of regular switch in Home app 😀

i3laze, 2021-11-07:
Надо в онлайн-конфигураторе выкинуть всё лишнее (например, уведомления между аксессуарами).

Может, завтра вычищу и пришлю.

nov-, 2021-11-13:
> Надо в онлайн-конфигураторе выкинуть всё лишнее (например, уведомления между аксессуарами).
> 
> Может, завтра вычищу и пришлю.

Все получилось, спасибо большое 😊