RavenSystem/esp-homekit-devices #440: HAA updates

xtremedays, 2019-06-22:
Just a quick question, is there a quick easy way to update the firmware on haa devices or do you have to go into setup mode?

i3laze, 2019-06-24:
Looks like:
1. Power cycle
2. 10x toggle via Home.app
3. Logging into your router to find correct IP-address DHCP lease
4. Logging into web interface of HAA
5. Checking OTA Update box + Save.

xtremedays, 2019-06-24:
That’s what I read too, but if they’re light switches behind a wall it’s not the easiest thing to cycle the power!

RavenSystem, 2019-06-28:
Actually, that is the only way to update HAA. Other methods will be added in the future. Remember that HAA is in a beta status, and it has only a few functions.

closed 2019-06-28 by RavenSystem

dwhp, 2019-06-28:
Hi José. 

May I suggest one future method to reset WiFi in HAA is to have the option to set a gpio dedicated to do so. Like hold for 10secs or whatever. Then we could add or use buttons as dedicated Wifi reset. 

Sorry to put this in the bottom of a closed issue and if I missed reference to idea prior.

RavenSystem, 2019-06-29:
@dwhp That option is one that will be in future versions.