RavenSystem/esp-homekit-devices #441: Th as basic

mafyata, 2019-06-22:
Hi , I have Sonoff TH16. Can I configure it in eve to work as Sonoff basic(without temp and humidity info, only switch)? Will it be safe for use ?

RavenSystem, 2019-06-22:
Of course. It is safe, and you will be able to manage 16A instead 10A. I used that setup with some devices to power 3500W without risks.

mafyata, 2019-06-23:
Board type 1 or 2 ?

RavenSystem, 2019-06-23:
That depends of your TH16 model. But board type only makes sense if you use an external toggle connected to sensor GPIO.

mafyata, 2019-06-23:
[image]
I think mine is newer model, I bought it May 2019

closed 2019-07-20 by RavenSystem