RavenSystem/esp-homekit-devices #442: Using non sonoff devices - ESP8266 based power strip

Nekomae, 2019-06-24:
Hello

Thanks for the amazing firmware!

I was wondering if it would be possible to use this firmware with a non sonoff ESP8266 based power strip. It's known as SM-SO301, koi or teepad power strip. 4 independent relay mains sockets and bank of 4 usb ports via single relay. 

Someone has it working with tasmota and has shared GPIO info. 

https://github.com/arendst/Sonoff-Tasmota/wiki/SM-SO301

I don't know much about programming but it seems like it might be possible to add as ESP8266 based and we know the GPIO info. It is also similar hardware to sonoff.

Could one of the profiles for sonoff be modified to work?

Thanks!

mafyata, 2019-06-25:
i am using this led controller for my led strip. It is not sonoff. I use ravencore with esp 8266 and esp 8285.
[image]

Nekomae, 2019-06-25:
very cool!

How did you got about doing it?

mafyata, 2019-06-25:
Same way you flash sonoff. Open the case of controller and solder cables. I will search for instructions for soldering.

mafyata, 2019-06-25:
[image]

mafyata, 2019-06-25:
Ground and IOO connect together.
You can read here:
https://github.com/arendst/Sonoff-Tasmota/wiki/MagicHome-LED-strip-controller

Nekomae, 2019-06-25:
ahh i see. The programming bit is no problem. 

Did you have to change the code at all?

mafyata, 2019-06-25:
Code?

Nekomae, 2019-06-25:
ahh i mean the ravenscore source code. Or did you just pick the nearest device listed in the firmware?

mafyata, 2019-06-25:
While flashing use the same code - esptool.py........ In eve app use device type 14

Nekomae, 2019-06-26:
ok. So you used the eve app to configure the GPIO for LED's. 

Can I use the same device type to turn on 4 power relays for each socket?

Nekomae, 2019-06-26:
Actually there are 5 relays. 

/Users/NekoNeko/Downloads/43324706-6d14d972-917a-11e8-9be3-b6f6f4ef4dcf.JPG

Nekomae, 2019-06-26:
Actually there are 5 relays. 

https://user-images.githubusercontent.com/3240875/43324706-6d14d972-917a-11e8-9be3-b6f6f4ef4dcf.JPG

mafyata, 2019-06-26:
[image]
[image]
This is for led strip in eve app. Your picture is not for ravencore. I think that you can’t use this device type for relays. See wiki page. There may find answer and device type for your relays.

Nekomae, 2019-06-26:
The closest device is the 4 ch switch. 

It appears that i can change the GPIO pin numbers in the source code and recompile it. 

I would still need to control one more relay. I don't know if this is possible to add in the code or not. 

This is what i was asking originally. Can the source code be modified to work with this power strip.

RavenSystem, 2019-06-26:
Use Home Accessory Architect firmware instead RavenCore.

Nekomae, 2019-06-26:
D'OH!

Reading the first paragraph of HAA it seems exactly what i need. 

Thanks for the help!

closed 2019-06-28 by RavenSystem