RavenSystem/esp-homekit-devices #446: Request new feature: triggering http urls

massimo79, 2019-07-04:
@RavenSystem 
Hi! Would it be possibile for HAA to implement a dummy switch/sensor to trigger an http request (ora a short bit.ly url)? Thanks, it would avoid many users to implement node.js servers like Homebridge and would be a really nice feature to add!

RavenSystem, 2019-07-04:
It is interesting.

massimo79, 2019-07-05:
@RavenSystem 
I can give you a couple of examples if you want. My a/c (Daikin) and my alarm panel (Visonic) have optional online controllers. Both of these can be DIY with esp8266 devices running native api’s.

The result is that they can be easily controlled via http calling  internal ip’s or even remotely via port forwarding.

Would be great to remote access via Homekit triggering the http commands, for example:

http://ip.address/armhome 
http://ip.address/armaway
http://ip.address/off

Furthermore they are always in sync even when controlled locally with ir remotes!!!

Tks! Keep up the great work!

i3laze, 2019-07-12:
Sorry for highjacking, I’ve got opposite request:
Allow HAA to be triggered (on/off/toggle) by http-GET request (via wget or curl).

Same reason: to get rid of HomeBridge.

noobydp, 2019-07-12:
@massimo79 could you link me to the daikin API info?

massimo79, 2019-07-12:
@noobydp check these out:
https://github.com/andreacucci/IR-Daikin-Server-ESP8266
https://github.com/ael-code/daikin-control

RavenSystem, 2020-01-16:
Featured added with v1.3.0.

closed 2020-01-16 by RavenSystem