RavenSystem/esp-homekit-devices #449: Tested device

egorist, 2019-07-06:
Hello. I bought esp device to control dc motor. I don't know it's [name,](https://aliexpress.com/item/32944121554.html) but it is working good with ravencore device 12 board type  2, external toggle 1. Maybe it helps someone.
[image]

[image]

RavenSystem, 2019-07-06:
Many thanks!!

egorist, 2019-07-07:
sorry, but is it possible to add this icon to device 12?
[image]

RavenSystem, 2019-07-20:
No, that is a different accessory type. In future, HAA will let you do what you want.

closed 2019-07-20 by RavenSystem