RavenSystem/esp-homekit-devices #451: Garagedoor inching

kafes33, 2019-07-13:
Hi,

Thank you for your amazing work!
I have a question about increase inching time to more than 12s in Sonoff sv - Garagedoor? is it possible?

Thank you Very much

RavenSystem, 2019-07-14:
No, sorry. 12s is the max value, and it is far enough to trigger any commercial garage door.

kafes33, 2019-07-14:
My garage door is Hormann - supramatic , it open / close in 15 s:( so, 12s it is too short.

skanico, 2019-07-14:
I have the same garage door. That is not inching time. Inching time is 0.3 - 0.6 sec (the time you maintain the push button to open the door). 15s is "working time" in case you don't have a closed switch.

kafes33, 2019-07-15:
I tried set it up in 3 sec and this was does not function.
Ok, I will try it 0.3 sec. Thanks

kafes33, 2019-07-15:
Hi, skanico, that's right! garage door is fully functions!
Thank you!

closed 2019-07-15 by kafes33