RavenSystem/esp-homekit-devices #452: sonoff basic "No response" status

3uperduck, 2019-07-14:
My homekit device has 1 sonoff basic @ 0.5.4, 6 sonoff basic @ 1.0.0 and with 2 homepods.
AP is 1 Cisco Aironet 3702i fat ap mode.

They're randomly no response happen when I open homekit, have to wait about 10~60sec or even more. I've no idea what's wrong. Btw, 2 homepods never happened.

I've tried to change other ap, re-flash the devices, isolate the device with 0.5.4 firmware but it just keeps happening. 

When devices show "no response" I ping it, it replies ping, and if I ping a good condition device around 3 or more mins, device show "no response" unreachable in homekit, but still reply ping, dame wired.

Does anyone know what to deal with?

Thanks.
[image]
[image]

RavenSystem, 2019-07-14:
Your network has a problem managing multicast DNS traffic.

3uperduck, 2019-07-15:
Ok, I'll change a router to try soon

RavenSystem, 2019-07-15:
Maybe it is a router configuration problem.

3uperduck, 2019-07-15:
[image]
Should be not, that's a simple, basic, ISP provided device.

RavenSystem, 2019-07-15:
Can you show wan/lan and wireless options?

3uperduck, 2019-07-15:
sure

for wan/lan on router, yeap, that's all.
[image]
for wireless
[image]

If need anything more just tell me, thanks

RavenSystem, 2019-07-15:
Try setting DTIM to 3, and enabling Reliable multicast to WGB. Remember to reset your router to ensure changes will be applied.

3uperduck, 2019-07-16:
Nothing change, I think maybe my router doesn't support mDNS? I borrowed another one from my friend. Let's see.

3uperduck, 2019-07-17:
It seems Cisco AP configuration has some problem. Yesterday I use original Cisco AP via netgear router I borrowed, it just work at fist few hours (around 2~3) than, all device randomly "No Response". Then I try AP on netgear's router, it works fine till now.

I'll try to use CLI to config more, thanks.

RavenSystem, 2019-07-18:
I hope you can solve the issue with your Cisco IOS.

closed 2019-07-20 by RavenSystem