RavenSystem/esp-homekit-devices #454: Support for Sonoff Mini

danielhelmstedt, 2019-07-15:
Hey, is it possible to get a profile for the new Sonoff Mini on RavenCore?

https://www.itead.cc/sonoff-mini.html

https://community.home-assistant.io/t/is-this-a-new-sonoff-device-sonoff-mini/119774/57

As per thread above - it uses GPIO4 for the external toggle switch

RavenSystem, 2019-07-15:
No, better use HAA.

danielhelmstedt, 2019-07-15:
Why no? There's already three different options for Sonoff board layouts, as well as Shelly (which this is pretty much a direct competitor to). Seems kinda weird to just not support this particular device?

The relay already works with the standard Sonoff board layout, the only thing that needs changing is the GPIO for the ext. switch

RavenSystem, 2019-07-15:
Because it is supported by Home Accessory Architect firmware.

danielhelmstedt, 2019-07-15:
Well yes, everything is supported by HAA. Couldn't that same logic be applied to every supported board layout? So then what's the point of having RavenCore at all? Why support existing Sonoff Basics and Shelly 1, but not the Sonoff Mini? This will easily replace Sonoff Basics for many people who have them installed inside light switches and other devices.

FWIW, I know it's easy enough to do with HAA (I've now got one working as intended, thank you :D ), I just don't see a good reason NOT to add this as an option in RC.

RavenSystem, 2019-07-15:
Because RavenCore was the first, but now with HAA, RavenCore will be replaced with it. And I don't want to spend more time adding devices to a firmware that will be removed when HAA does all what RC does... and more.

danielhelmstedt, 2019-07-16:
Ahh okay, sorry, I didn’t realise RC was going to be phased out. Thanks for your awesome work on this!

closed 2019-07-16 by danielhelmstedt