RavenSystem/esp-homekit-devices #456: Sonoff SV issue

Jakub1403, 2019-07-16:
Hi!

I have a Sonoff SV with RavenCore software and I have a problem. I use it for my garage door.

A few days ago I changed my Wifi router and I wanted to change wifi settings in Sonoff

I don't know what's I did wrong but now my Sonoff doesn't work properly :(

when I connect power, after a few seconds one green LED blinks (3-4 times) and nothing else

when I push a little switch on Sonoff board green LED blinks 4 times

when I turn off my WiFi after about 5mins Sonoff appears in available networks and I can connect to it (192.168.4.1) and then I choose another network, write the password and click join (or connect - I don't remember)

I try to add it to Home in my iPhone and it doesn't appear

I try to push and hold a little switch on the Sonoff board (5, 10, ...,60 seconds) - nothing happens

What is wrong? How can I repair it? Should I re-flash RavenCore (I have to wait because I don't have any USB-TTL)?

Have you ever a similar problem?

Thanks in advance

Jacob

fcesc, 2019-07-17:
Hi,

I had a similar issue and I solved it flashing blank and reflashing.

Francesc.

Jakub1403, 2019-07-17:
Thanks! I’ll try to reflash later this week - I’m waiting for USB-TTL programmer

ArielRi, 2019-07-17:
You can do it with a Raspberry PI, do you have it?

Jakub1403, 2019-07-17:
I have a Raspberry Pi zero. I think tomorrow I’ll get TTL adapter so I’ll wait. Thanks for reply

closed 2019-08-25 by RavenSystem