RavenSystem/esp-homekit-devices #458: AP mode issue

pilot1981, 2019-07-21:
Hi,
I have Shelly 1 with RC. It worked fine for about 2 months.
Now It is not responding in HomeKit and I see WIFI network Ravencore-xxxx, but my iPhone isn’t able to connect to this network.

What I have to do to solve this issue?

Regards

St0nehead, 2019-07-21:
Try to choose Ravencore-xxxx in you Settings/WiFi, then press Forget This Network.
Now try again to connect to Ravencore-xxxx

pilot1981, 2019-07-21:
I don’t have this option: iPhone don’t remember last time it connect to this SSID...

pilot1981, 2019-08-08:
issue disappeared itself; now shelly is again connected to my home kit installation...very strange

pilot1981, 2019-08-11:
Today I  had the same problem with 2 sonoff basic in same position:

After power outage, I see 2 SSID RAVENCORE-xxxx, so for one of them I connected to WiFi and I inserted again WIFI settings, for the other one I had to reboot several times...

Any idea?

peros550, 2019-08-18:
@RavenSystem I would like to ask for your advice here.  

Correct me if I am wrong, I think there are occasions in which Ravencore's version of  of esp-wifi-config module will turn the device into AP mode indefinitely. 

When router SSID is not available, Ravencore will look for it's known router SSID and try to connect for only 5 minutes (WIFI_CONFIG_CONNECT_TIMEOUT=300000)  before it gives up and turns into AP mode.   

The problem is that when in AP mode, the device does not try to connect again with the router's SSID. It just sits there and wait for client connections.  

5 minutes is good choice to cover the startup time for the majority of routers but there can be various reasons why router Wifi became unavailable for more than 5 mins (router firmware updates, isolated room power cuts, some people use scheduled operation of Wifi for whatever reasons , etc).

The new implementation of esp-wifi-config module has resolved this issue. Even if the device turns into AP mode, it keeps looking for the router's SSID and will connect as soon as it finds it again as available. 

Looking forward for your view on the above. 

Thanks,

RavenSystem, 2019-08-25:
Fixed. Now device will search for wifi network forever.

closed 2019-08-25 by RavenSystem