RavenSystem/esp-homekit-devices #459: No response from Sonoff basic

joshuahare17, 2019-07-21:
Hey guys. I’ve got a fair few sonoffs around the house al running the ravencore software. What I’m finding is that 2 of them seem to go unresponsive, this includes the manual press button on them. The only way to bring them back online is to reset the power. These 2 particular sonoffs are also controlled a lot through the Apple HomeKit automations for timers, could this be causing them to crash? Any help would be appreciated.

fcesc, 2019-07-22:
I have  2 sonoff basic working with the Apple HomeKit automations timer and work fine.

Partizan7676, 2019-07-28:
it may be problems with wi-fi. what equipment you use?

PS first i recommend erase and re-flash them

mientki, 2019-08-01:
I have same problem :(

xtremedays, 2019-08-05:
Tried using different sonoffs on the same circuits? could just be faulty units. all 16 of mine work all good with plenty of automations etc running.

mientki, 2019-08-05:
on all my 4 items is the same problem.  wifi I have quite strong unifi ac lite.

djdom111, 2019-08-29:
Hi Guys, I have the same problem as you. I have 9 Sonoff Basic devices. Before the Ravencore firmware I used them with tasmota and homebridge. Everything was fine, when I changed to Ravencore 3 of them have this issue. Randomly are unresponsive. At the beginning I thought that it is a problem with WiFi signal. But one of it is 2 meters from my AC router. Do you know any solution for this problem? I tried to reset and flash a new firmware. Nothing helps.

pilot1981, 2019-08-29:
This is WiFi issue: try to change WIFI ROUTER/AP!

djdom111, 2019-08-29:
I does look like WiFi issue, but how to explain that on tasmota and homebridge using the same router was everything fine?

peros550, 2019-08-29:
Guys, this is a very special issue. It does not happen to everyone, therefore a lot of different reasons  may be causing it. We need to provide as much as possible Information on our setups (ie router type, how many devices, how often the issue happens, ping test result when the issue occurred, use of WiFi extenders, troubleshoot steps that worked/not worked, version of ravencore, custom code or not ) that’s just a few of information points I can think of. If José agrees, an organized approach could be followed , by creating a stick issue with a questionnaire like the above so that everyone can provide their responses and try to find common denominator(s)

djdom111, 2019-08-29:
All my Sonoff devices are on the latest firmware (changing the firmware 4 times since I’ve setup it for the first time didn’t change anything). My router is Asus RT-AC87U. What I can tell more about the issue are:
1. It happens couple times every hour
2. The sonoff device with issue blinking with a green light often compare to the “good” ones.

closed 2019-11-07 by RavenSystem

honza67, 2020-01-07:
> 
> 
> I have 2 sonoff basic working with the Apple HomeKit automations timer and work fine.

Hi, does SONOFF basic still work on automation (timer or switch via motion detector)? For me about a month state that the SONOFF elements have stopped working through automation and can be controlled only manually via phone, tablet or MacBook in the application, but if they are in some automatic mode, so do not switch (does not change the state). Before that, it normally worked for about a year. Very strange condition ....

RavenSystem, 2020-01-07:
@honza67 what firmware and version are you using?

honza67, 2020-01-08:
**@ RavenSystem** Firmware I have Gruppio - Sonoff OFF.bin

RavenSystem, 2020-01-08:
@honza67 Sorry, but what are you doing here? You should ask it to the developer of your firmware.

honza67, 2020-01-08:
**@RavenSystem** sorry, but I try to find out if this problem is not on uncertified fw in general .. probably there could be some change in Apple.

RavenSystem, 2020-01-08:
@honza67 You must use Gruppio - Sonoff OFF.bin developer repo to ask this.