RavenSystem/esp-homekit-devices #465: Wifi changed and homekit does not find the devices anymore ...

RajeeRah, 2019-08-02:
Hi RavenSytem Team,

Thank you for your great job, the devices i am using with you software works really good and performs very well.

But now I moved to our new home and so I have new Network with new Wifi settings. So i had to reset the devices to change the wifi.
1) i was able to reset
2) i was able to find the devices in wifi
3) i was able to change the new Wifi and when i check Router, there are also those devices
4) Problem: I tried to add the devices to homekit, but its just loading and loading and nothing happens.

any tip for me to solve this isseu? am I doing something wrong?
do i have to reflash everytime?

Thank you in advance. Have a nice weekend.

i3laze, 2019-08-08:
Why didn’t you just rename your WiFi network to the old SSID?

Not only RC devices but all your iPhones and stuff should remember new credentials. Too much hustle)

RajuRah, 2019-08-09:
> Why didn’t you just rename your new WiFi network to the old SSID+pass?
> Not only RC devices but all your iPhones and stuff have to remember new credentials - Too much hustle)
> 
> Try deleting devices from Home.app and resetting the by rapidly pressing a button 10x times.

The old SSID was a random one and we moved recently and now i am setting up the home network proper.
- the option to changing the SSID on shelly was one of the reason why I flashed with RavenSystem.
- some of my friends wants also shelly which i have to flash, so it will be great if they can change SSID as well. 

I try again to delete in home.app and reset shelly with 10 x times. (I tried also few times and will do today few more time lol ... same issue also with home magic led strip controller)

i3laze, 2019-08-09:
https://github.com/RavenSystem/esp-homekit-devices/wiki/RavenCore

> Resetting device to factory defaults
> Within first minute since device boots, wait about 5 seconds to allow device to boot completely, and then toggle any switch quickly (from Home App or from external switches) at least 10 times.

RajuRah, 2019-08-09:
alright, I will try the second part you mentioned and let you know, how it went. Thanks!

RajuRah, 2019-08-19:
> https://github.com/RavenSystem/esp-homekit-devices/wiki/RavenCore
> Here's the main way, so you shoudn't actually delete them from Home.app:
> 
> > Entering Wifi AP mode
> > When a power cycle occurs, device will try to connect to its known wifi network. If after 5 minutes wifi network is unreachable, device will enter in AP mode so you can connect to it and reconfigure wifi settings.
> 
> Indeed sometimes there are pairing issues after ssid change. If bunch of powercycles won't help to complete the pairing sucessfully, only then remove & reset:
> 
> > Resetting device to factory defaults
> > Within first minute since device boots, wait about 5 seconds to allow device to boot completely, and then toggle any switch quickly (from Home App or from external switches) at least 10 times.

Hi, so finally I had time this weekend to check the issue.
1) Entering Wifi AP mode when device can not find Wifi for 5 min - did not work with homekit
    - i was able to see the device in wifi list
    - i was able to type the new wifi and PW
    - i was able to see the device in the router with MAC and IP addresses
    - BUT HOMEKIT was not able to find it

2) RESETTING to Factory defaults - worked fine on Sonoff basic and Shelly 1
    - Shelly 1 was completely wired (including with standard switch)
    - I had to turn off the Switch 10 times (like in WIKI)

3) Now the next problem i have is, I have a Magic Home RGB controller which has same issue
    - I CAN NOT RESET TO FACTORY DEFAULTS - Because the Controller does not have manual switch (like sonoff or the stanrdard house switch for shelly 1)
    - Can i use GPIO-0 and GRN to reset it ???? 

Thank you for the help.

RavenSystem, 2019-08-19:
Can i use GPIO-0 and GRN to reset it ????
Yes. 

> El 19 ago 2019, a las 13:00, RajuRah <notifications@github.com> escribió:
> 
> Can i use GPIO-0 and GRN to reset it ????

RajeeRah, 2019-08-20:
> Can i use GPIO-0 and GRN to reset it ???? Yes.
> […](#)
> El 19 ago 2019, a las 13:00, RajuRah ***@***.***> escribió: Can i use GPIO-0 and GRN to reset it ????

Superb, it works fine, thank you very much. I have a RGBW strip, which GPIO number is for the white? currently its mix with 3 Colors. I would like to donate your great work guys, where can I do this?

Thanks again.

i3laze, 2019-08-20:
> I would like to donate your great work guys, where can I do this?

A nice button on GitHub title page:
https://paypal.me/ravensystem

closed 2019-11-07 by RavenSystem